Jpe230/SonicPad-Debian #1: Some remarks

nofuturekid, 2023-06-09:
I know you said no support, so I will just inform you that in your image some things are missing.

1. Somehow only Klipper and Moonraker are installed but no Mainsail etc. as you mentioned.
2. Additionally maybe a good  addition would be to make use of the blue LED again with putting something like
`echo "mmc0" /sys/class/leds/sys-led/trigger  # Blink blue led when mmc0 is accessed`
into your rc.local

Jpe230, 2023-06-09:
Thanks!

>Somehow only Klipper and Moonraker are installed but no Mainsail etc. as you mentioned.

I removed the preinstalled Mainsail from the image at last second, since, ultimately it is a personal decision to choose the preferred front-end :) I've removed the mentions of Mainsail in the README. Thanks for letting me know.

> Additionally maybe a good addition would be to make use of the blue LED again

I can it add to the misc. section, I would like to leave the functionality of the LED as an exercise to the reader :)

nofuturekid, 2023-06-10:
Sure, fair enough.

Maybe there is another issue. You changed the partition layout and there is a file `/etc/fw_env.config`.
In there it specifies where `fw_printenv` and `fw_setenv` are expecting the environment variables for UBoot.
Looking at:
```shell
sonic@sonicpad:~$ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0   7.3G  0 disk
├─mmcblk0p1  179:1    0     4M  0 part
├─mmcblk0p2  179:2    0   512K  0 part
├─mmcblk0p3  179:3    0   512K  0 part
├─mmcblk0p4  179:4    0    30M  0 part
├─mmcblk0p5  179:5    0   6.9G  0 part /
└─mmcblk0p6  179:6    0 301.5M  0 part
mmcblk0boot0 179:8    0     4M  1 disk
mmcblk0boot1 179:16   0     4M  1 disk
mmcblk0rpmb  179:24   0     4M  0 disk
```

I would say it should be:

```shell
# Configuration file for fw_(printenv/setenv) utility.
# Up to two entries are valid, in this case the redundant
# environment sector is assumed present.
# Notice, that the "Number of sectors" is not required on NOR and SPI-dataflash.
# Futhermore, if the Flash sector size is omitted, this value is assumed to
# be the same as the Environment size, which is valid for NOR and SPI-dataflash
# Device offset must be prefixed with 0x to be parsed as a hexadecimal value.

# Sonic Pad
/dev/mmcblk0p2                          0x0000  0x20000
/dev/mmcblk0p3                          0x0000  0x20000
```

Jpe230, 2023-06-10:
Thanks I've addressed this in the lastest release,

I've added the config used under [scripts/fw_env.config](https://github.com/Jpe230/SonicPad-Debian/blob/main/scripts/fw_env.config)

As well as example services (with your suggestion for the led status) to deprecate `rc.local`  ^.^

closed 2023-06-15 by nofuturekid

marcelolcosta, 2023-06-21:
Hi,
Thank you so much for your contribution. I installed the OS yesterday and everything is working. 
I remember seeing a topic about the configuration of the g-sensor, but I cant find it anymore. could you, please, point it to me?

Cheers,