Jpe230/SonicPad-Debian #4: G-sensor configuration

marcelolcosta, 2023-06-21:
Hi,
Thank you so much for your contribution. I installed the OS yesterday and everything is working.
I remember seeing a topic about the configuration of the g-sensor, but I cant find it anymore. could you, please, point it to me?

Cheers,

marcelolcosta, 2023-06-21:
answering to my own question:
1 - cd ~/klipper
2 - sudo cp ./scripts/klipper-mcu.service /etc/systemd/system/
3 - sudo systemctl enable klipper-mcu.service
4 - make menuconfig
5 - select "Linux process" as microcontroller architecture
6 - save and exit
7 - sudo service klipper stop
8-  make flash
9 - sudo service klipper start
10 - add to printer.cfg:
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev2.0

[resonance_tester]
accel_chip: adxl345
probe_points:
    122, 115, 10
```

closed 2023-06-21 by marcelolcosta

RyanDevineD, 2024-05-14:
I have tried to follow the steps in this Issue, but I still can't get the original ADXL to work with the Sonic Pad now it has been flashed with Klipper. When I add the config I am told about an issue with rpi, any ideas?

AdamSLevy, 2024-07-12:
> answering to my own question:
> 1 - cd ~/klipper
> 2 - sudo cp ./scripts/klipper-mcu.service /etc/systemd/system/
> 3 - sudo systemctl enable klipper-mcu.service
> 4 - make menuconfig
> 5 - select "Linux [...]

Worth mentioning that after step 2, you might need to run `systemctl daemon-reload` for systemd to become aware of the new `klipper-mcu.service`. Step 3, enabling the unit may fail otherwise. Alternatively a reboot works.

Note also that enabling a service does not start it at that time, just enables it to be run on boot.