Jpe230/SonicPad-Debian #7: Klipperscreen screen rotation

iLbsLi, 2023-07-01:
Hi,
I tried to flip the screen as to the Klipperscreen documentation, could you please maybe add a simple script to make it easier. thanks

Jpe230, 2023-07-04:
Confirm working using `fbdev` values for rotation, ie:
```
The supported values are "CW" (clockwise, 90 degrees), "UD" (upside down, 180 degrees) and "CCW" (counter clockwise, 270 degrees). Implies use of the shadow framebuffer layer. Default: off.
```
My xorg conf looks like this:
```
Section "Device"
        Identifier      "Allwinner R818 FBDEV"
        Driver           "fbdev"
        Option         "Rotate" "UD"
        Option         "fbdev" "/dev/fb0"
        Option         "SwapbuffersWait" "true"
EndSection
```
and to rotate the touch input:
```
DISPLAY=:0 xinput set-prop 7 'Coordinate Transformation Matrix' -1 0 1 0 -1 1 0 0 1
```

closed 2023-07-04 by Jpe230

iLbsLi, 2023-07-04:
> Confirm working using `fbdev` values for rotation, ie:
>
> ```
> The supported values are "CW" (clockwise, 90 degrees), "UD" (upside down, 180 degrees) and "CCW" (counter clockwise, 270 degrees). Implies [...]

that worked beautifully, you're the best  :)