Jpe230/SonicPad-Debian #8: Accelerometer Problem

Printo1987, 2023-07-02:
Hello Can I use this same accelerometer from sonic pad? 
I install everything and this pacckages:
sudo apt install avrdude gcc-avr binutils-avr avr-libc stm32flash binutils-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib gcc-arm-none-eabi

add printer:
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev2.0

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points:
  110,110,20

but nothing;/ and can't connect with printer if I add this below to printer.cfg

Everything working but not accelerometer. Maybe I must enable SPI? but how?

nofuturekid, 2023-07-03:
You must "flash" Klipper also on the host. That's the "[mcu rpi]" thing... see here: https://www.klipper3d.org/RPi_microcontroller.html

Printo1987, 2023-07-03:
> You must "flash" Klipper also on the host. That's the "[mcu rpi]" thing... see here: https://www.klipper3d.org/RPi_microcontroller.html

Super Working :) Thanks

closed 2023-07-03 by Jpe230