Jpe230/SonicPad-Debian #10: No MCU

Printo1987, 2023-07-05:
My Debian Sonic pad can't find mcu I plug in by USB:

I using by Kiauh get mcu and see:
No MCU found!
MCU either not connected or not detected!

nofuturekid, 2023-07-05:
Can you see your printer/mainboard with using 'lsusb' or 'dmesg'?

Printo1987, 2023-07-05:
before I using rpi3 and normal working but I change this sonic for Delta becuase I wanna using accelometr with Sonic and NO MCU;/

nofuturekid, 2023-07-05:
At least there is no known general problem. Without further details it is hard to help.

So please provide more information, like what have you tried and ie. the output of already mentioned commands.

Printo1987, 2023-07-05:
if I check MCU UART by kiauh i see : MCU #1: ttyS0

but I connected my mainboard by USB

nofuturekid, 2023-07-05:
Please provide requested information, otherwise I can't help.

Printo1987, 2023-07-05:
I install everything if I go kiauh GET MCU  i see NO MCU;/

if I plugin my previously RPI3  i see: 
serial: /dev/serial/by-id/usb-Klipper_sam3x8e_203120514D3652323133303035303234-if00

I've  Arduino Due and on this I've compilation flash by USB

nofuturekid, 2023-07-05:
I need the requested information to help you and you are not providing them. Sorry, I'm out here then.

Printo1987, 2023-07-05:
> I need the requested information to help you and you are not providing them. Sorry, I'm out here then.

I must chec this right?
Can you see your printer/mainboard with using 'lsusb' or 'dmesg'? 

after copy and paste what i see on the screen?

Printo1987, 2023-07-05:
If i use by putty comend:
sonic@SonicPad:~$ lsusb
-bash: lsusb: command not found

if I use dmesg i see: 

sonic@SonicPad:~$ dmesg
[    2.515530] hub 1-1:1.0: USB hub found
[    2.516118] hub 1-1:1.0: 4 ports detected
[    2.545151] IR RC5(x/sz) protocol handler initialized
[    2.550842] IR RC6 protocol handler initialized
[    2.555949] IR JVC protocol handler initialized
[    2.561052] IR Sony protocol handler initialized
[    2.566255] IR SANYO protocol handler initialized
[    2.571554] IR Sharp protocol handler initialized
[    2.576853] IR MCE Keyboard/mouse protocol handler initialized
[    2.583421] IR XMP protocol handler initialized
[    2.589601] usbcore: registered new interface driver uvcvideo
[    2.596079] USB Video Class driver (1.1.1)
[    2.600698] sunxi cedar version 0.1
[    2.604896] VE: install start!!!

[    2.610434] cedar_ve: cedar-ve the get irq is 344
[    2.616068] VE: line 1811 set the sram data

[    2.623403] VE: get debugfs_mpp_root is NULL, please check mpp

[    2.631640] VE: sunxi ve debug register driver failed!

[    2.639094] VE: install end!!!

[    2.646299] Bluetooth: HCI UART driver ver 2.2.74e8f89.20201026-111523
[    2.653665] Bluetooth: HCI UART protocol (null) registered
[    2.659851] Bluetooth: HCI H4 protocol initialized
[    2.665249] Bluetooth: XRadio Bluetooth LPM Mode Driver Ver 01.00.07
[    2.672975] bt_fdi debugfs_init
[    2.686921] sunxi-mmc sdc2: SD/MMC/SDIO Host Controller Driver(v3.48 2020-9-1                                      5 16:01)
[    2.695961] sunxi-mmc sdc2: ***ctl-spec-caps*** 8
[    2.701917] sunxi-mmc sdc2: No vdmmc regulator found
[    2.707516] sunxi-mmc sdc2: No vd33sw regulator found
[    2.713208] sunxi-mmc sdc2: No vd18sw regulator found
[    2.718903] sunxi-mmc sdc2: No vq33sw regulator found
[    2.724591] sunxi-mmc sdc2: No vq18sw regulator found
[    2.731220] sunxi-mmc sdc2: set host busy
[    2.735834] mmc:failed to get gpios
[    2.740525] sunxi-mmc sdc2: sdc set ios:clk 0Hz bm PP pm UP vdd 22 width 1 ti                                      ming LEGACY(SDR12) dt B
[    2.769800] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm PP pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.797809] sunxi-mmc sdc2: detmode:alway in(non removable)
[    2.797830] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm PP pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.806214] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm PP pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.807302] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm OD pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.818165] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm OD pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.818248] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm OD pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.831107] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm OD pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.876278] sunxi-mmc sdc2: sdc set ios:clk 400000Hz bm PP pm ON vdd 22 width                                       1 timing LEGACY(SDR12) dt B
[    2.887299] sunxi-mmc sdc1: SD/MMC/SDIO Host Controller Driver(v3.48 2020-9-1                                      5 16:01)
[    2.887440] sunxi-mmc sdc1: ***ctl-spec-caps*** 8
[    2.887581] sunxi-mmc sdc1: No vmmc regulator found
[    2.887585] sunxi-mmc sdc1: No vqmmc regulator found
[    2.887590] sunxi-mmc sdc1: No vdmmc regulator found
[    2.887594] sunxi-mmc sdc1: No vd33sw regulator found
[    2.887599] sunxi-mmc sdc1: No vd18sw regulator found
[    2.887604] sunxi-mmc sdc1: No vq33sw regulator found
[    2.887608] sunxi-mmc sdc1: No vq18sw regulator found
[    2.888505] sunxi-mmc sdc1: set host busy
[    2.888612] mmc:failed to get gpios
[    2.889393] sunxi-mmc sdc1: sdc set ios:clk 0Hz bm PP pm UP vdd 21 width 1 ti                                      ming LEGACY(SDR12) dt B
[    2.889417] sunxi-mmc sdc1: no vqmmc,Check if there is regulator
[    2.905801] sunxi-mmc sdc1: sdc set ios:clk 400000Hz bm PP pm ON vdd 21 width                                       1 timing LEGACY(SDR12) dt B
[    2.925807] sunxi-mmc sdc1: detmode:manually by software
[... 297 lines cut ...]
[    9.034241] [STA] !!!xradio_vif_setup: id=0, type=2, p2p=0
[    9.048358] [AP_WRN] BSS_CHANGED_ASSOC but driver is unjoined.
[    9.072500] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   14.833591] systemd-journald[971]: Failed to set ACL on /var/log/journal/b2a6                                      c3288af54222b0308c96c71df261/user-1000.journal, ignoring: Operation not supporte                                      d
[   14.925634] PVR_K:  1727: RGX Firmware image 'rgx.fw.22.102.54.38' loaded
[   62.846306] random: crng init done
[   62.850109] random: 7 urandom warning(s) missed due to ratelimiting
[   69.152147] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   70.757807] [STA_WRN] !!! xradio_remove_interface: vif_id=0
[   70.766572] [STA] !!!xradio_vif_setup: id=0, type=2, p2p=0
[   70.780466] [AP_WRN] BSS_CHANGED_ASSOC but driver is unjoined.
[   70.804270] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   72.409230] [STA_WRN] Freq 2437 (wsm ch: 6) prev: 3.
[   72.414838] wlan0: authenticate with 18:34:af:96:04:7a (try 1)
[... cut ...]
[   72.473733] [TXRX_WRN] [RX] IF=0, No key found.
[   72.481585] [STA_WRN] Freq 2437 (wsm ch: 6) prev: 3.
[   72.487173] wlan0: associate with 18:34:af:96:04:7a (try 1)
[   72.499465] wlan0: RX AssocResp from 18:34:af:96:04:7a (capab=0x1411 status=0                                       aid=20)
[   72.508244] wlan0: associated
[   72.513719] [AP_WRN] [STA] ASSOC HTCAP 11N 58
[   72.531028] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   72.564273] [TXRX_WRN] drop=1440, fctl=0x00d0.
sonic@SonicPad:~$

nofuturekid, 2023-07-05:
Okay, fine. Plug in the Printer and run "dmesg" again. This time only the last 20 to 30 lines of the output should be sufficient.

nofuturekid, 2023-07-05:
I'm not sure why you don't have 'lsusb' installed, but you could and should install it via: 'sudo apt-get install usbutils' because also some Klipper utilities depend on that.

Printo1987, 2023-07-05:
> I'm not sure why you don't have 'lsusb' installed, but you could and should install it via: 'sudo apt-get install usbutils' because also some Klipper utilities depend on that.

lsusb:

sonic@SonicPad:~$ lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dmesg:

[    9.102408] usb usb3: Manufacturer: Linux 4.9.191 ehci_hcd
[    9.108565] usb usb3: SerialNumber: sunxi-ehci
[    9.115844] hub 3-0:1.0: USB hub found
[    9.120170] hub 3-0:1.0: 1 port detected
[    9.126210] [ohci0-controller]: sunxi_usb_enable_ohci
[    9.131907] [sunxi-ohci0]: probe, pdev->name: 5101000.ohci0-controller, sunxi                     _ohci: 0xffffff8008b29550
[    9.142799] sunxi-ohci 5101000.ohci0-controller: SW USB2.0 'Open' Host Contro                     ller (OHCI) Driver
[    9.152677] sunxi-ohci 5101000.ohci0-controller: new USB bus registered, assi                     gned bus number 4
[    9.162511] sunxi-ohci 5101000.ohci0-controller: irq 354, io mem 0xffffff8008                     552614
[    9.234390] usb usb4: New USB device found, idVendor=1d6b, idProduct=0001
[    9.241980] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=                     1
[    9.250053] usb usb4: Product: SW USB2.0 'Open' Host Controller (OHCI) Driver
[    9.258064] usb usb4: Manufacturer: Linux 4.9.191 ohci_hcd
[    9.264233] usb usb4: SerialNumber: sunxi-ohci
[    9.271532] hub 4-0:1.0: USB hub found
[    9.275858] hub 4-0:1.0: 1 port detected
[   14.867091] systemd-journald[970]: Failed to set ACL on /var/log/journal/b2a6                     c3288af54222b0308c96c71df261/user-1000.journal, ignoring: Operation not supporte                     d
[   14.967294] PVR_K:  1726: RGX Firmware image 'rgx.fw.22.102.54.38' loaded
[   63.378034] random: crng init done
[   63.381840] random: 7 urandom warning(s) missed due to ratelimiting
[   69.151495] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   70.765875] [STA_WRN] !!! xradio_remove_interface: vif_id=0
[   70.774544] [STA] !!!xradio_vif_setup: id=0, type=2, p2p=0
[   70.788459] [AP_WRN] BSS_CHANGED_ASSOC but driver is unjoined.
[   70.812204] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   72.421932] [STA_WRN] Freq 2437 (wsm ch: 6) prev: 3.
[   72.427563] wlan0: authenticate with 18:34:af:96:04:7a (try 1)
[   72.479450] wlan0: authenticated
[   72.483147] [STA_WRN] Freq 2422 (wsm ch: 3) prev: 6.
[   72.488835] [STA_WRN] Freq 2437 (wsm ch: 6) prev: 3.
[   72.494448] wlan0: associate with 18:34:af:96:04:7a (try 1)
[   72.504732] wlan0: RX AssocResp from 18:34:af:96:04:7a (capab=0x1411 status=0                      aid=35)
[   72.513509] wlan0: associated
[   72.518894] [AP_WRN] [STA] ASSOC HTCAP 11N 58
[   72.536603] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   72.583478] [TXRX_WRN] drop=1440, fctl=0x00d0.
sonic@SonicPad:~$

nofuturekid, 2023-07-05:
So from what I see is that the pad does not recognise your printer at all. Btw, have already tried different USB ports?

Printo1987, 2023-07-05:
> So from what I see is that the pad does not recognise your printer at all. Btw, have already tried different USB ports?

Yeap I try all 4 port usb and nothing
I try too conect for my 2nd printer Voron switchwire  and this same NO MCU;/

Original Klipper Sonic working normally with VOron and Delta but I prefer this Debian is better and  fresh Klipper.

nofuturekid, 2023-07-05:
Okay then, I don't know how, but it looks like something went wrong during installation and as you state that with SonicOS it was working, I assume your USB ports are fine.

So I would suggest to re-flash the Debian Image, as this is the fastest option. After a successful flash I would check the working condition of the USB before doing anything else.

with the given commandos of 'lsusb' and 'dmesg' - there should be changing output when a printer is connected and disconnected again.

Printo1987, 2023-07-05:
> W porządku, nie wiem jak, ale wygląda na to, że coś poszło nie tak podczas instalacji, a ponieważ twierdzisz, że z SonicOS działało, zakładam, że twoje porty USB są w porządku.
> 
> Sugerowałbym więc ponowne flashowanie obrazu Debiana, ponieważ jest to najszybsza opcja. Po udanym flashowaniu sprawdziłbym stan działania USB przed zrobieniem czegokolwiek innego.
> 
> z podanymi komendami 'lsusb' i 'dmesg' - po podłączeniu i ponownym odłączeniu drukarki powinny występować zmiany wyjścia.

very much thanks for your time and help ... I dont know whats going on :(  I reinstall too new Build Debian Sonic and nothing;/

Jpe230, 2023-07-06:
The kernel is the same as the stock SonicPadOS, so if the stock os recognized it, it should work with Debian.

What's the output of?:
```
ls /dev/serial/by-id/
```

Printo1987, 2023-07-06:
Guys Working:)

Yeasterday I change 3 cable includes cable from Sonic and no working;/  Today I find shortly cable plug to my mainboard and find MCU:)

Very thanks for fast contact and help:)

@Jpe230 you can close :)

closed 2023-07-06 by Jpe230