Jpe230/SonicPad-Debian #9: Storage expantion

iLbsLi, 2023-07-05:
Hi,

As the sonic pad have 8GB it got almost full or at least that's what it's telling me when I try to install anything now, even thought I installed only Klipper, Moonraker, Mainsail, Klipperscreen & Crowsnest!
So is that normal? and is there a way around it? is there a way to expand the storage with usb drive?
keep up the good work.
thanks

nofuturekid, 2023-07-05:
Sadly it's pretty normal because of a lot of additional software which is needed for compiling Klipper.
This in theory could be removed afterwards and might free up additional space again, but once in a while you would need it again, because for compiling Klipper for flashing onto your printer again.

But you are free to use a USB stick/disk to expand your storage and mount it anywhere in the system and tell Klipper/Moonraker to use this then for storing your GCodes.

Jpe230, 2023-07-06:
As mentioned above you can remove unused packages and reinstall them later when needed. Mounting a usb and using it for Goode should be possible as well. 

Closing this issue since I can't really do something about disk space

closed 2023-07-06 by Jpe230