Jpe230/SonicPad-Debian #11: Storage issue

iLbsLi, 2023-07-07:
Hi,

I know I'm opening an issue for the same problem but, my system reached a point where it can't update anything what out giving me a fail for no  space, even though I don't any g-code files stored, so I'm reflashing to Debian again to see if that fixes the problem, at least temporarly and my question is could it be possible to run debian from the usb storage? if not debian could I install klipper and other stuff on the usb storage? and how please 😅

iLbsLi, 2023-07-07:
So I flashed to Debian again the it was successful (./phoenixsuit t800-sonic-debian.img) and system seemed to restart to a fresh install, I connected to wifi and tried to update klipper and it failed to low space! something not right here. what am I missing?!

nofuturekid, 2023-07-07:
Looks like you missed the step of increasing the file system size.
Please re-read the instructions.

After flashing and before updating you have to resize the filesystem to the 8GB.

iLbsLi, 2023-07-07:
> Looks like you missed the step of increasing the file system size. Please re-read the instructions.
> 
> After flashing and before updating you have to resize the filesystem to the 8GB.

Wow, thanks, I actually never did that as I have never followed the instructions :p  that explais why I was so low in space for now apparent reason I'll check it out now ;)

iLbsLi, 2023-07-07:
Problem solved with a single line 😅, Thanks to nofuturekid, although I'm still interested to know if there's an answer to my main questions.

nofuturekid, 2023-07-07:
In theory almost everything is possible. What isn't possible by now is booting from USB, as this would need support from the UBOOT bootloader. This would mean to have to recompile the bootloader. But one could put almost anything else onto a USB stick to save space on the internal EMMC.

closed 2023-07-18 by Jpe230