Jpe230/SonicPad-Debian #12: How to Configure ADXL?

beroliv, 2023-07-28:
is there a Manual how to  configure the ADXL with your image? would be nice i could still use it.

nofuturekid, 2023-07-28:
Just connect it and put the correct / same config as for the Sonic Pad with Creality OS into your "printer.cfg"

beroliv, 2023-07-28:
i upload my backup from the sonic pad to the new flashed debian. i get also a unknown command measure_axes_noise in the screen

nofuturekid, 2023-07-28:
Just follow the official Klipper documentation, as this is nothing different than any installation installation of Klipper with Kiauh.

beroliv, 2023-07-28:
there is also a locale problem.. because make menuconfig are not working also,
in the Printer CFG the adxl was outcommentet with # so i remove them but now i have no connection to the printer

make menuconfig
Loaded configuration '/home/sonic/klipper/.config'
Traceback (most recent call last):
  File "/home/sonic/klipper/lib/kconfiglib/menuconfig.py", line 3281, in <module>
    _main()
  File "/home/sonic/klipper/lib/kconfiglib/menuconfig.py", line 661, in _main
    menuconfig(standard_kconfig(__doc__))
  File "/home/sonic/klipper/lib/kconfiglib/menuconfig.py", line 705, in menuconfig
    locale.setlocale(locale.LC_ALL, "")
  File "/usr/lib/python3.9/locale.py", line 610, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
make: *** [Makefile:116: menuconfig] Error 1

nofuturekid, 2023-07-28:
That's not usual and should not happen. Not sure what you have done.

Usually flash the OS, expand the filesystem to full size and be happy.

beroliv, 2023-07-28:
i work on a mac but this fix dont work for me
https://github.com/Klipper3d/klipper/issues/3669

beroliv, 2023-07-28:
input shaper should work

nofuturekid, 2023-07-28:
That has nothing to do with Mac.

Flash the Sonic Pad again with the Debian OS, expand the filesystem as explained on the Download/Releases Page and from there on follow official Klipper documentation.

This is basically like any Klipper installation on a Raspberry Pi with Kiauh.

beroliv, 2023-07-28:
make menuconfig is working now. i had to set the locales because i loggin first with a Macos

sudo dpkg-reconfigure locales i check en_GB.UTF-8... and
  en_US.UTF-8... 

sudo locale-gen

beroliv, 2023-07-28:
klipper MCU are not starting ? 

klipper-mcu.service - Starts the MCU Linux firmware for klipper on startup
     Loaded: loaded (/etc/systemd/system/klipper-mcu.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
  Condition: start condition failed at Fri 2023-07-28 22:06:46 CEST; 2min 55s ago
             └─ ConditionFileIsExecutable=/usr/local/bin/klipper_mcu was not met
       Docs: https://www.klipper3d.org/RPi_microcontroller.html

beroliv, 2023-07-28:
i do the steps in the manual again and now it works.. that was a bit sketchy with the locales.. 

thank you for you effort making this image and help!!

nofuturekid, 2023-07-28:
Ohh, no, no... it's not mine. I'm just a random guy who knows a bit about Linux, Klipper and accidentally was the first one trying this thing besides the originator.

beroliv, 2023-07-28:
im als knowing a bit about linux otherwise you are very lost with problems like that. i try to calibrate now my printer and after that a test print. happy to get i work. i was a bit annoyed about the pad after i find the bad things out.

beroliv, 2023-07-28:
i have to hit retry until it connects to the printer.. need longer then the timeout time..

closed 2023-07-30 by beroliv