Jpe230/SonicPad-Debian #13: Klipperscreen update

beroliv, 2023-07-30:
Klipperscreen shows not in Moonraker update Manger up!

i add the following lines in moonraker.conf

[update_manager KlipperScreen]
type: git_repo
path: ~/KlipperScreen
origin: https://github.com/jordanruthe/KlipperScreen.git
env: ~/.KlipperScreen-env/bin/python
requirements: scripts/KlipperScreen-requirements.txt
install_script: scripts/KlipperScreen-install.sh

Jpe230, 2023-08-25:
The scripts to build includes a function to patch the moonraker configuration: https://github.com/Jpe230/SonicPad-Debian/blob/1f0300de21d84ddd9f0a65abfcf347763ef180ae/src/scripts/install_klipperscreen.sh#L33

The prebuilt image doesn't have it but it would be a good idea create a new release with the lastest changes :)

Jpe230, 2024-03-05:
Closing this issue since it is now fix in the latest release!

closed 2024-03-05 by Jpe230