Jpe230/SonicPad-Debian #15: Kiauh telegram bot install failure

gjm19761, 2023-08-01:
i getv this when trying to  install telegram bot

Building wheels for collected packages: Pillow, ujson, wsaccel, tornado
  Building wheel for Pillow (setup.py) ... done
  Created wheel for Pillow: filename=Pillow-9.4.0-cp39-cp39-linux_aarch64.whl size=1066748 sha256=47a9e81fd4aac5ad30a7f7f44fd4e9bf72b1e7529466ca0e1da0a477c29e9ef3
  Stored in directory: /home/sonic/.cache/pip/wheels/a5/c4/5f/0df28f27675dec97dd2bc428577bf912202dd10791f9deefd1
  Building wheel for ujson (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/sonic/moonraker-telegram-bot-env/bin/python /tmp/tmpasbw_v8k_in_process.py build_wheel /tmp/tmpz3xnynir
       cwd: /tmp/pip-install-8nqv_es4/ujson_50f10889178844ae938e09f21cf3baff
  Complete output (126 lines):
  running bdist_wheel
  running build
  running build_ext
  creating build
  creating build/temp.linux-aarch64-3.9
  creating build/temp.linux-aarch64-3.9/deps
  creating build/temp.linux-aarch64-3.9/deps/double-conversion
  creating build/temp.linux-aarch64-3.9/deps/double-conversion/double-conversion
  creating build/temp.linux-aarch64-3.9/lib
  creating build/temp.linux-aarch64-3.9/python
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/bignum-dtoa.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/bignum-dtoa.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/bignum.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/bignum.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/cached-powers.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/cached-powers.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/double-to-string.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/double-to-string.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/fast-dtoa.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/fast-dtoa.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/fixed-dtoa.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/fixed-dtoa.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/string-to-double.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/string-to-double.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./deps/double-conversion/double-conversion/strtod.cc -o build/temp.linux-aarch64-3.9/./deps/double-conversion/double-conversion/strtod.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./lib/dconv_wrapper.cc -o build/temp.linux-aarch64-3.9/./lib/dconv_wrapper.o -D_GNU_SOURCE
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./lib/ultrajsondec.c -o build/temp.linux-aarch64-3.9/./lib/ultrajsondec.o -D_GNU_SOURCE
  ./lib/ultrajsondec.c:69:47: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
     69 | static JSOBJ FASTCALL_MSVC decode_any( struct DecoderState *ds) FASTCALL_ATTR;
        |                                               ^~~~~~~~~~~~
  ./lib/ultrajsondec.c:80:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
     80 | {
        | ^
  ./lib/ultrajsondec.c:92:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
     92 | {
        | ^
  ./lib/ultrajsondec.c:250:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    250 | {
        | ^
  ./lib/ultrajsondec.c:270:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    270 | {
        | ^
  ./lib/ultrajsondec.c:292:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    292 | {
        | ^
  ./lib/ultrajsondec.c:312:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    312 | {
        | ^
  ./lib/ultrajsondec.c:363:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    363 | {
        | ^
  ./lib/ultrajsondec.c:590:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    590 | {
        | ^
  ./lib/ultrajsondec.c:654:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    654 | {
        | ^
  ./lib/ultrajsondec.c:748:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    748 | {
        | ^
  arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-jS0VHk/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I./python -I./lib -I./deps/double-conversion/double-conversion -I/home/sonic/moonraker-telegram-bot-env/include -I/usr/include/python3.9 -c ./lib/ultrajsonenc.c -o build/temp.linux-aarch64-3.9/./lib/ultrajsonenc.o -D_GNU_SOURCE
  ./lib/ultrajsonenc.c:183:1: warning: ‘fastcall’ attribute directive ignored [-Wattributes]
    183 | {
        | ^
[... cut ...]
  Building wheel for tornado (setup.py) ... done
  Created wheel for tornado: filename=tornado-6.1-cp39-cp39-linux_aarch64.whl size=421722 sha256=4f6d8d496b1b5fc360d46cf07298a357572cc3be3823072a170b2c1c55e165f9
  Stored in directory: /home/sonic/.cache/pip/wheels/e6/6b/c3/dbb71bdabdc4681fb3cdf364f2c969f39ebc13c1a512cf6e43
Successfully built Pillow wsaccel tornado
Failed to build ujson
ERROR: Could not build wheels for ujson which use PEP 517 and cannot be installed directly
sonic@SonicPad:~/kiauh$

Jpe230, 2023-08-25:
Sorry for the late response, can you try with the latest version of KIAUH?

Locally I can install the telegram bot without issues:

[image]

Jpe230, 2023-08-26:
One solution is to use piwheels to download the precompiled for dependencies. Create the file "/etc/pip.conf" and add the contents of this file:
https://github.com/Jpe230/SonicPad-Debian/blob/main/src/base_rootfs/etc/pip.conf

closed 2023-10-29 by Jpe230