Jpe230/SonicPad-Debian #16: Changing Boot Splash Image, M300 Support?

mdewyer, 2023-08-04:
Been using this for the past week on my SP and it's been great.

Any ideas on how to:

1. Change the boot splash image from the Debian logo to something else?
2. Use the Pad's internal speaker for M300's?

Thanks!

Jpe230, 2023-08-06:
1) I haven't looked into changing the splash screen with the firmware already flashed; under "boot_resource" there's a bmp with the default (debian) splash screen, you can try changing it and creating a flashable image, alternative the splash screen is saved into a partition, it should be writable within Linux, see if you can update it there, only consideration is the available space within the partition 

2) totally untested, the stock kernel has traces for an audio driver (alsa) but I haven't use it or use it

mdewyer, 2023-08-06:
Good deal - thanks for your insight, and for putting the project together!

closed 2023-08-06 by mdewyer