Jpe230/SonicPad-Debian #18: How do I rotate the display?

BakeJake, 2023-08-20:
I'd love to use this but with my current configuration I cant have the screen in its original orientation, needing to flip it but couldn't figure out how to do it.

Jpe230, 2023-08-21:
https://github.com/Jpe230/SonicPad-Debian/issues/7#issuecomment-1620587643

closed 2023-08-21 by Jpe230

Jpe230, 2023-08-21:
> > [#7 (comment)](https://github.com/Jpe230/SonicPad-Debian/issues/7#issuecomment-1620587643)
> 
> @Jpe230 
> 
> Where do I put this?
> 
> `DISPLAY=:0 xinput set-prop 7 'Coordinate Transformation Matrix' -1 0 1 0 -1 1 0 0 1`
> 
> Nvm, it says to run in terminal on klipperscreen docs, thanks!

That's a command, you can enter it manually in the terminal or a script at boot-time (rc.local or systemd)

BakeJake, 2023-08-21:
> That's a command, you can enter it manually in the terminal or a script at boot-time (rc.local or systemd)

@Jpe230 How would I make that boot script?

BakeJake, 2023-08-21:
> That's a command, you can enter it manually in the terminal or a script at boot-time (rc.local or systemd)

@Jpe230, 
What fixed it for me was the bottom part of this with the values for flipping instead of those "-1 0 1 0 -1 1 0 0 1" and the device id of 7:
[image]