Jpe230/SonicPad-Debian #22: Creality Serial to USB adapter.

SaltyNewfie, 2023-09-03:
I'm trying to get the connection via the adapter but it requires a driver from what I gather, I've tried installing it but this build is missing the  build folder in lib/modules. And I can't seem to find the headers for the kernel as it's deprecated.

nofuturekid, 2023-09-03:
Are you sure it needs an additional driver? Usually Linux Kernel has such basic stuff like USB to serial compiled in. What needs to be changed for sure is your configuration (printer.cfg) and the mainboard needs to be flashed again, to reflect the change.

Jpe230, 2023-09-04:
> I'm trying to get the connection via the adapter but it requires a driver from what I gather, I've tried installing it but this build is missing the build folder in lib/modules. And I can't seem to find the headers for the kernel as it's deprecated.

Hard to know without having a dump of the updated kernel/image from Creality, the kernel source used by my port is located here: https://github.com/Jpe230/SonicPadOS/tree/debian you can try to see if a kernel module is missing from the build.

maksii, 2023-09-07:
@SaltyNewfie  I'm on "Sonic Pad Debian Bullseye" and [official cable](https://www.creality3dparts.com/product/creality-sonic-pad-serial-cable/) works fine for me without any additional steps required. 
Just update firmware and printer.cfg to match the new mcu serial path for it.

Or are you using some other usb-to-uart cable?

Jpe230, 2023-09-09:
> @SaltyNewfie  I'm on "Sonic Pad Debian Bullseye" and [official cable](https://www.creality3dparts.com/product/creality-sonic-pad-serial-cable/) works fine for me without any additional steps [...]

Glad to know that it works, a little off-topic but what's the use for this cable?

maksii, 2023-09-10:
> little off-topic but what's the use for this cable?

In a nutshell, just personal preference.
In theory, of course, there are advantages over the usb, but in practice they will not be noticeable.

Personally, I use it because of the ability to run the wire out to either side of the printer and get rid of the 5v problem from the usb, so you can completely turn off the printer with a smart outlet.
Many users use it simply because of the more stable connection (since the usb connector may not hold the cable as well over time). [Lost communication with MCU](https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors)

mdewyer, 2023-09-11:
While we're on the subject.. does anyone happen to have the serial firmware that the printer gets flashed with? Running the stock Creality OS (if you will) the process seems to be: factory reset the SonicPad and go through the setup initial process again and choose serial connection, and it'd save the firmware file(s) to a thumb drive that you then plug into the printer to update.

Since I'm no longer running the stock Creality code I no longer can go through that process and get those files, and haven't been able to find them anywhere online.

If anyone is able to obtain and share the file(s) that'd be super helpful, not just to me but to others who may find themselves in a simlar situation after switching to this project. 

In my case I'm running a 4.2.2 board.

Thanks!

maksii, 2023-09-11:
@mdewyer just follow official Klipper documentation and make new firmware(this project readme contains details on missing libs, etc). you may want to use winscp\filezilla(or other software) to simplify the copy of bin file later on.

mdewyer, 2023-09-11:
> @mdewyer just follow official Klipper documentation and make new firmware(this project readme contains details on missing libs, etc). you may want to use winscp\filezilla(or other software) to simplify the copy of bin file later on.

Oh! My apologies! I didn't realize it wasn't Creality-specific. Thank you!

maksii, 2023-09-11:
> Oh! My apologies! I didn't realize it wasn't Creality-specific. Thank you!
This project just "gives" you full access to OS(Debian) on SonicPad(eg, mini-pc with touch screen) with out of the box klipper setup.

So any steps\action related to configuration, setup, customization or printer settings should be followed as per official [Klipper](https://www.klipper3d.org/) documentation. 
There are some specifics due to how hardware builded and sources are provided(eg limitation with reboot, some libs, storage, etc), but other than that - any guide 'how to install klipper on creality 4.2.2' will help you.

if for some reason, you forgot to save your config before moving to this project - you can find one that will work for you [here](https://github.com/Klipper3d/klipper/tree/master/config). they also contains details of what options should be selected during firmware make

Gavrik13, 2023-10-18:
Hi!
I couldn’t connect the printer to the sonicpad, either via USB or through a native adapter.
The klipper simply wouldn't connect to the printer.
Having rolled back to the native software from Creality, everything connected the first time... :(
I suspect that the problem is in the firmware configuration of the printer microcontroller. But I can't understand why...

Gavrik13, 2023-10-19:
> just flash the original Firmware after that. Flash the Klipper Firmware you will see it works

that's exactly what I did. But the problem is that having, for example, only raspberrie, it’s impossible to deliver a klipper

beroliv, 2023-10-19:
do you create the rigth firmware that fits on your printer? the Processor Model?

Gavrik13, 2023-10-19:
> do you create the rigth firmware that fits on your printer? the Processor Model?

stm32 F401

beroliv, 2023-10-19:
you create it exactly like that?

[image]

Gavrik13, 2023-10-19:
> you create it exactly like that? Bootloader and Comunication
> 
Yes, I tried both advanced and simple modes.
both UART and USB.

closed 2024-06-19 by Jpe230