Jpe230/SonicPad-Debian #23: using adxl gives me problems

Egalistel, 2023-09-10:
Hello, I am looking for information on the use of the adxl provided as standard because I do not understand how to make it work?

I followed the instructions carefully, I sent the lines of code via ssh... but after that? on the screen I have the macro, but no adxl detected...
On the original image, I had lines concerning it, but it no longer seems to be valid? I may have missed something, if someone could enlighten me? this is one of the reasons why I changed the image because on my Core X/Y the sensor made the pad spit during the test.

In any case, congratulations for the work.
Thanking you.

maksii, 2023-09-10:
Just follow the official documentation. There are no changes to this project that affect how it works.
https://www.klipper3d.org/Measuring_Resonances.html#software-installation

Egalistel, 2023-09-11:
> Just follow the official documentation. There are no changes to this project that affect how it works. https://www.klipper3d.org/Measuring_Resonances.html#software-installation

Thank you for your response, having some problems understanding English, I did not correctly understand the installation principle.
It works now, I just noticed that I couldn't have the configuration activated in my 2 machines at the same time.

I am therefore stuck on one last thing, the resonance test crashes on my core :)

Thanks again

closed 2023-10-02 by Jpe230