Jpe230/SonicPad-Debian #25: Add ip_tables to kernel

PhilippF1992, 2023-09-24:
Hi, 
First of all thanks for this repo. 
I got a Sonic Pad but I stopped using it for my 3D printer and thought about reusing it as a Input for my HomeAssistant installation. 

That is working fine. I changed your scripts to not include Klipper, Moonraker and Klipperscreen but install chromium and start it via a service in fullscreen mode. That is working fine and I got no problems with that. 

Now I thought about using the pad as a homeassistant instance in my motor home (together with espHome and maybe mqtt) and wanted to install docker to run homeassistant supervised. 

But I cannot even run docker. 
It complains about "ip_tables" not being part of the kernel. 

I am new to building and compiling kernels and I do not even know where to start or what to properly google. Can you help me with this? Maybe point me in the right direction and I can keep going from there. 

I guess there will be more problems coming down the line and I do not even know if the pad is able to handle the workload. But I feel like this is a nice start to enter the "kernel-world" and learn a lot! 

Thanks in advance.

nofuturekid, 2023-09-25:
Hi Philipp,
sounds like cool idea, as I'm using HA myself I'm also interested in such. Sadly it's not that easy to build a newer kernel for the pad. So we will have to stick with the original kernel. But Creality has published the sources, so it should be possible to follow Creality's original instructions on building the SonicPad OS and just strip out the kernel from there. Obviously modifying it to include 'iptables'.

Performance wise the SonicPad should be able to handle this, as it's nearly as capable as a Raspberry Pi.

Maybe I find some time today to have a look into Creality's guide and can up with at least some links to get you started in compiling their delivered Kernel sources.

Jpe230, 2023-10-02:
As mentioned above, you will need to recompile the kernel to enable such feature, my personal fork used in this project is located here: https://github.com/Jpe230/SonicPadOS/tree/debian.

And the configuration for the kernel is located here: https://github.com/Jpe230/SonicPadOS/blob/debian/device/config/chips/r818/configs/sonic_lcd/linux/config-4.9.

I'm closing this issue for now, since it doesn't really fall under the project use case, but I'll try to update this repo in how to recompile and replace the kernel in case a user needs a feature currently disabled.

Cheers.

closed 2023-10-02 by Jpe230