Jpe230/SonicPad-Debian #31: USB stick and webcam not working

AK3DLv, 2023-10-08:
As the title says.
Having an issue with getting the Sonic pad to read the USB stick when sticking into any other port except the one I'm using for connecting to the printer.
Also, installed Crowsnest for monitoring via IP of the print, but it can't seem to find the webcam.
The webcam in question is a Logitech P270. Has anyone else had this issue and somehow solved it?

m4sis, 2023-12-01:
the same I'm looking but I can't find how to make the webcam work

AK3DLv, 2023-12-06:
> the same I'm looking but I can't find how to make the webcam work

I plugged in webcam and installed Crowsnest via KIAUH and it worked. Still have no idea how to make USB stick to work.

sfeole, 2024-01-23:
> > the same I'm looking but I can't find how to make the webcam work
> 
> I plugged in webcam and installed Crowsnest via KIAUH and it worked. Still have no idea how to make USB stick to work.

Hey There, 
As I understand your problem, Only 1 USB Port on your Sonic Pad is working with your USB Stick.

I can help you debug,

SSH into your sonic pad:
```
#ssh sonic@IP
#sudo apt update && sudo apt install usbutils
```
```
sonic@SonicPad:~$ sudo lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

You should see the USB Hub Devices.

Let's watch the dmesg live

```
$sudo dmesg -w
```

*Plug in your USB Stick in the back of the Sonic Pad. (I used the port marked *USB*)

```
[56123.520874] usb 1-1.2: new high-speed USB device number 4 using sunxi-ehci
[56123.671079] usb 1-1.2: New USB device found, idVendor=125f, idProduct=312b
[56123.678804] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[56123.687035] usb 1-1.2: Product: ADATA USB Flash Drive
[56123.692862] usb 1-1.2: Manufacturer: ADATA
[56123.697971] usb 1-1.2: SerialNumber: 1510611512340057
[56123.707110] usb-storage 1-1.2:1.0: USB Mass Storage device detected
[56123.715260] scsi host0: usb-storage 1-1.2:1.0
[56125.724707] scsi 0:0:0:0: Direct-Access     ADATA    USB Flash Drive  0.00 PQ: 0 ANSI: 6
[56125.738127] sd 0:0:0:0: [sda] 30344192 512-byte logical blocks: (15.5 GB/14.5 GiB)
[56125.747852] sd 0:0:0:0: [sda] Write Protect is off
[56125.753541] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
[56125.760905] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[56125.787352]  sda: sda1
[56125.796114] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

USB Stick is identified as */dev/sda*
```
sonic@SonicPad:~$ sudo lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 004: ID 125f:312b A-DATA Technology Co., Ltd. Superior S102 Pro
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

sonic@SonicPad:~$ sudo lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda            8:0    1 14.5G  0 disk 
`-sda1         8:1    1 14.5G  0 part 
mmcblk0      179:0    0  7.3G  0 disk 
|-mmcblk0p1  179:1    0    4M  0 part 
|-mmcblk0p2  179:2    0  512K  0 part 
|-mmcblk0p3  179:3    0  512K  0 part 
|-mmcblk0p4  179:4    0   30M  0 part 
|-mmcblk0p5  179:5    0  7.2G  0 part /
`-mmcblk0p6  179:6    0  512B  0 part 
mmcblk0boot0 179:8    0    4M  1 disk 
mmcblk0boot1 179:16   0    4M  1 disk 
mmcblk0rpmb  179:24   0    4M  0 disk 
```

Now, at this point I can successfully tell you that my USB Stick is correctly identified and is working.  
* Have you tried the following above?  
* What is it your trying to accomplish?

closed 2024-03-05 by Jpe230