Jpe230/SonicPad-Debian #33: Same Wifi MacAdresses on 2 Devices

cyclemat, 2023-10-14:
today i have flash a seconf pad but the problem is now the twodevices have the same mac adress so the DHCP trys to give booth the same ip adress and cant handle two pads at the same time. has anybody two pads and confirm this ?

MAcadress on my pads is fc : ee : 92 : 11 : 14 : 05

nofuturekid, 2023-10-15:
I only have just 1 pad but I can confirm that this is the exact same MAC address mine is using.

It can be 'reconfigured' somewhere in the "/etc" folder.

cyclemat, 2023-10-15:
have test macchanger but didnt work !

nofuturekid, 2023-10-15:
I'm currently too lazy to dig more into it but this, while it's for orange pi, looks promising: https://dietpi.com/forum/t/orange-pi-zero-wlan-mac-address/929

cyclemat, 2023-10-15:
have test to change the macadress via macchanger

sudo systemctl stop moonraker.service
sudo systemctl stop KlipperScreen.service
sudo ip link set dev wlan0 down
sudo ip link set dev eth0 down
sudo macchanger -r wlan0
sudo macchanger -r etho
sudo ip link set dev wlan0 up
sudo ip link set dev eth0 up
sudo systemctl start moonraker.service
sudo systemctl start KlipperScreen.service

on eth0 it works but on the wlan0 

>  wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 5e:25:53:fe:da:08 brd ff:ff:ff:ff:ff:ff

nofuturekid, 2023-10-15:
Maybe this:
`
cat /etc/modprobe.d/xradio_wlan.conf

options xradio_wlan macaddr=DC:44:6D:D3:A6:7F
`

Check existence of that file and content. If it not exists, create it. Reboot.

cyclemat, 2023-10-15:
nope didnt work 
[image]
[image]

nofuturekid, 2023-10-15:
I'm nowhere near my pad. So I can't try out some things myself.
But tomorrow I will.

cyclemat, 2023-10-15:
have found this https://dietpi.com/forum/t/orange-pi-zero-change-mac-wifi/1028/2

MAC must be changed in file /etc/udev/rules.d/70-persistent-net.rules
MAC must be changed in /etc/modprobe.d/xradio_wlan.conf

but he didnt write what must in 70-persistent-net.rules

nofuturekid, 2023-10-16:
For me it's working as expected

```
sonic@sonicpad:~$ ip a

4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether fc:ee:91:00:29:ef brd ff:ff:ff:ff:ff:ff
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b2:06:ab:26:bf:90 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether fc:ee:91:00:29:ee brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.28/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863784sec preferred_lft 863784sec

sonic@sonicpad:~$ sudo grep -iR "fc:ee:91:00:29:ee" /etc/
/etc/wifi/xr_wifi.conf:FC:EE:91:00:29:EE
grep: /etc/runit/runsvdir/default/ssh/log/supervise: No such file or directory
grep: /etc/runit/runsvdir/default/ssh/supervise: No such file or directory
grep: /etc/sv/ssh/log/supervise: No such file or directory
grep: /etc/sv/ssh/supervise: No such file or directory
```

Then I was simply running:

```
echo "FC:EE:91:00:29:EF" | sudo tee /etc/wifi/xr_wifi.conf
sudo reboot
```

And when the Sonic Pad was available again:

```
ip a
4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether fc:ee:91:00:29:f0 brd ff:ff:ff:ff:ff:ff
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b2:06:ab:26:bf:90 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether fc:ee:91:00:29:ef brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.25/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 863944sec preferred_lft 863944sec
```

BUT, I admit my system was already modified and I can't remember what all I did. One thing though which comes into my mind is:

```
cat /etc/rc.local 
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

#fsck -p /dev/mmcblk0p5
#mount -o remount, rw /

echo "mmc0" > /sys/class/leds/sys-led/trigger	# Blink blue led when mmc0 is accessed
cat /sys/devices/platform/soc/usbc0/usb_host	# Enable CAM port as usb host
/usr/bin/brightness -v 25			# Reduce brightness a bit

exit 0
```

cyclemat, 2023-10-18:
`sonic@sonicpad:~$ echo "FC:EE:91:00:29:EF" | sudo tee /etc/wifi/xr_wifi.conf
sonic@sonicpad:~$ sudo reboot`

Works best was we create a first run script that starts and create the xr_wifi.conf with a randome mac adress.

qwazwsx, 2025-01-27:
Static mac address is provided by this file

https://github.com/Jpe230/SonicPad-Debian/blob/eaf2eefdb334714e77303a362c332bbc6eb3e55d/src/base_rootfs/etc/wifi/xr_wifi.conf

I think randomizing this at first boot would be beneficial