Jpe230/SonicPad-Debian #34: Printer not found

Scipolah, 2023-10-15:
I'm having the same problem that Presto1987 had, my sonic pad doesn't find the printer. Can't see it either with lsusb or dmesg
Yes, I've tried different cables and different ports. If you need I can send any kind of logs.

nofuturekid, 2023-10-15:
1. Boot the pad without any printer connected
2. Connect via SSH
3. Attach printer via USB
4. Run 'dmesg | tail -n 10' and paste output
5. Run 'lsusb' and paste output

Scipolah, 2023-10-15:
> 1. Boot the pad without any printer connected
> 2. Connect via SSH
> 3. Attach printer via USB
> 4. Run 'dmesg | tail -n 10' and paste output
> 5. Run 'lsusb' and paste output

~$ dmesg | tail -n 10
[   58.003358] [AP_WRN] [STA] ASSOC HTCAP 11N 58
[   58.021050] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   58.088785] [TXRX_WRN] drop=1440, fctl=0x00d0.
[   58.422921] [TXRX_WRN] drop=1440, fctl=0x00d0.
[  118.232350] usb 1-1.2: new full-speed USB device number 3 using sunxi-ehci
[  118.349737] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523
[  118.357443] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber                       =0
[  118.365620] usb 1-1.2: Product: USB Serial
[  118.372639] ch341 1-1.2:1.0: ch341-uart converter detected
[  118.383363] usb 1-1.2: ch341-uart converter now attached to ttyUSB0

~$ lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

nofuturekid, 2023-10-15:
So you are using the USB to Serial cable from Creality? It's correctly recognized, see 'ch341-uart converter' line (last one from 'dmesg' output)

So you have to use '/dev/ttyUSB0' in your printer.cfg as serial address.

Scipolah, 2023-10-15:
I'm not using the serial cable, I'm using different kinds of usb-a to usb-c cables and also the stock one from creality.
The only things that i have plugged are power cable and cable to connect to printer

Scipolah, 2023-10-15:
i"ve also tried to use that adress but it doesn't work either

nofuturekid, 2023-10-15:
What is it you connected at around "118.xxx" seconds? Because this is clearly not the "usual" way a printer mainboard is detected.

As said, this looks like a USB-Serial converter.

Scipolah, 2023-10-15:
i think the normal cable usb-c to usb-a cable
i will retry from the start and send you the log

Scipolah, 2023-10-15:
~$ dmesg | tail -n 10
[   58.477901] [AP_WRN] [STA] ASSOC HTCAP 11N 58
[   58.495241] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   58.580181] [TXRX_WRN] drop=1440, fctl=0x00d0.
[   59.027570] [TXRX_WRN] drop=1440, fctl=0x00d0.
[  103.937727] usb 1-1.2: new full-speed USB device number 3 using sunxi-ehci
[  104.055214] usb 1-1.2: New USB device found, idVendor=1a86, idProduct=7523
[  104.062915] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber                                                                                                                                                             =0
[  104.071121] usb 1-1.2: Product: USB Serial
[  104.078123] ch341 1-1.2:1.0: ch341-uart converter detected
[  104.088713] usb 1-1.2: ch341-uart converter now attached to ttyUSB0

this time i've used original sonic pad cable connected it in the usb port in the back above the cam one, should be number 3

Scipolah, 2023-10-15:
dmesg | tail -n 10
[   59.861366] [AP_WRN] [STA] ASSOC HTCAP 11N 58
[   59.879693] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   59.969919] [TXRX_WRN] drop=1440, fctl=0x00d0.
[   60.506674] [TXRX_WRN] drop=1440, fctl=0x00d0.
[   79.565907] usb 1-1.3: new full-speed USB device number 3 using sunxi-ehci
[   79.683647] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523
[   79.691386] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[   79.699579] usb 1-1.3: Product: USB Serial
[   79.706732] ch341 1-1.3:1.0: ch341-uart converter detected
[   79.717386] usb 1-1.3: ch341-uart converter now attached to ttyUSB0

different port,  i think port n1, same cable

Scipolah, 2023-10-15:
Okay I don't know what i did but after re-flashing debian, updating everything and using "/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0" as mc port it started working
thank you for your help

closed 2023-10-15 by Scipolah