Jpe230/SonicPad-Debian #36: installing from mac

fridaystreet, 2023-11-26:
I realise this isn't really in your wheelhouse, but if anyone is able to provide any assistance on getting the pheonixtool to run on a mac that would be greatly appreciated.

Followed the instructions, but it just sits on the ---enter sem_wait---- screen. I'm assuming it just isn't able to detect the sonic pad. I'm running an M2 mac (silicon) only have ubs-c ports. Have a usb-c to usb cable plugged in, Not sure if that is causing the issue. Not sure why, it works with all other usb devices, but who knows.

Thanks in advance.

beroliv, 2023-11-26:
I have a intel imac and did it try. My Ender 3 is Dead since a while. So why not try to reflash the Pad back to Original.
I Used a USB C to USB C Cable wtih two adapters to USB A and Connect it to the Mac and the Cam Port on the Pad. The Cable i orderet thats time for the Pad i dont find it anymore :-/
I Download the Phoenix tools and the Image from Creality Github.
I Bring the Pad in the Flash Mode and start the Flash.
The bigest hassle was the Macs security Policy. The Mac blocks some dll and other files to load. After i allow it to load all this files it flashes the Pad with its Original Firmware like a Charm. I need a feew trys untli all files where loaded.

`Oliver@iMac PhoenixSuit_MacOS_T800 % ./phoenixsuit t800-sonic_lcd_uart0-1.0.*.img
*************Allwinnertech************
AW MacOS Burn firmware software V1.0.0
*************Allwinnertech************
-----LoadPluginCenter------- 
library file path: /Users/Oliver/Downloads/PhoenixSuit_MacOS_T800/plgvector.dll
library file path: /Users/Oliver/Downloads/PhoenixSuit_MacOS_T800/LiveProc.Plg
-----SetLiveProc------- 
library file path: /Users/Oliver/Downloads/PhoenixSuit_MacOS_T800/plgvector.dll
library file path: /Users/Oliver/Downloads/PhoenixSuit_MacOS_T800/luaeFex.dll
-----OnImage------- 
DeviceThreadEntryForMac called
Register./luaBase.dll l_RegAllFun Sucess!
Register./luaeFex.dll l_RegAllFun Sucess!
Register./luadec.dll l_RegAllFun Sucess!
copy luatool from local
----enter sem_wait------- 
./AllWinnerDevice.cpp Fex_Open 604
Found allwinner usb device (vendor 7994, product 61416), location id 14100000
Found 1 devices
Begin Burn the image! 
progress 0%Register./luaBase.dll l_RegAllFun Sucess!
Register./luaeFex.dll l_RegAllFun Sucess!
Register./luadec.dll l_RegAllFun Sucess!
copy luatool from local
./AllWinnerDevice.cpp Fex_Open 604
Found allwinner usb device (vendor 7994, product 61416), location id 14100000
Found 1 devices
./AllWinnerDevice.cpp Fex_Open 604
Found allwinner usb device (vendor 7994, product 61416), location id 14100000
Found 1 devices
--------------Init Called------------------
workDir	/Users/Oliver/Downloads/PhoenixSuit_MacOS_T800
ImgLenHigh	0
ImgLenLow	921485312
Mode	8
hWnd	0
imgFilePath	t800-sonic_lcd_uart0-1.0.6.43.51.img
[TL_MSG]:Mode = 8, ImgLenHigh=0, ImgLenLow = 36ecc000, imgFilePath = t800-sonic_lcd_uart0-1.0.6.43.51.img

IMAGEWTY
ItemTableSize = 1048576
[TL_MSG]:Tools Open Img

---fun end---
FEL RecordDev: 14100000
--------------entry-fel2fes Called-----------
felDevName	14100000
[TL_MSG]:Hi, I'm fel, dev=14100000

snip
`

fridaystreet, 2023-11-27:
thanks @beroliv yeah once you said about the cable I kicked myself as I had used a cable which was charging without thinking. But I've tried again with 2 other usb-c to usb-a cables which I know are data, in fact one of them is the cable I connect the sonic pad to my ender 3 s1 with. Still no joy. Had to go through all the security stuff, not getting any of those errors anymore. It looks like it loads the firmware utility firm and is waiting to connect to the device.

Even tried running my terminal application in rosetta. Can't get it to do anything with this custom firmware or the creality firmware on my M@. My wife has an intel mac, I'll give that a try tonight

nofuturekid, 2023-11-27:
You also made sure that you have started the program BEFORE connecting the Sonic pad?

fridaystreet, 2023-11-27:
pretty sure I did, but I'll try again. Also someone else mentioned to check the usb device is actually there. So might doa lsusb and see if it is being detected.

Jpe230, 2023-11-30:
My experience with my MacBook (Intel):
- Mac -> C - C cable -> passive usb adapter (C to A) -> Sonic Pad: Doesn't work
- Mac -> C - A cable -> Sonic Pad: Doesn't work
- Mac -> USB hub -> A - A cable -> Sonic Pad: Works 

Also the order in which the program is ran matter

Nocentinia, 2024-01-22:
sorry but have you found the .img file? because I download it but I only have one folder and the file I need isn't there. what should we do?

sfeole, 2024-01-23:
> sorry but have you found the .img file? because I download it but I only have one folder and the file I need isn't there. what should we do?

@Nocentinia I replied to your [bug](https://github.com/Jpe230/SonicPad-Debian/issues/42)

thevolcanomanishere, 2024-02-18:
I figured out the Mac connection problems. I had to made my own USB-A to USB-A cable to make it work. 

I have a MacBook Pro M1 btw.

You need USB - to USB hub that has an A socket. Then create a male to male with some spare USB cables. When you cut them open, you want to see 4 cables ideally. Join them together, make sure they don't short by wrapping them individually with tape