Jpe230/SonicPad-Debian #40: im wondering how to get the adxl345 working im getting Unknown pin chip name 'rpi' error in printer.cfg

Glod76, 2023-12-23:
Hello. I'm wondering how to get the adxl345 working I'm getting Unknown pin chip name 'rpi' error in printer.cfg
Otherwise loving this over the stock pad

chaoschris, 2024-01-03:
Have you installed the needed software packages as it is described in the documentation?
https://www.klipper3d.org/Measuring_Resonances.html#software-installation

Software installation

Note that resonance measurements and shaper auto-calibration require additional software dependencies not installed by default. First, run on your Raspberry Pi the following commands:

sudo apt update
sudo apt install python3-numpy python3-matplotlib libatlas-base-dev

Next, in order to install NumPy in the Klipper environment, run the command:

~/klippy-env/bin/pip install -v numpy

Note that, depending on the performance of the CPU, it may take a lot of time, up to 10-20 minutes. Be patient and wait for the completion of the installation. On some occasions, if the board has too little RAM the installation may fail and you will need to enable swap.

nofuturekid, 2024-01-03:
You could look up in the already closed issues.

0xdreadnaught, 2024-01-21:
The provided steps didn't work with the SPI sensor that comes with the pad. I had to do the following in addition to get it functional:

apt-get install spi-tools
chown sonic:sonic /dev/spidev2.0

#printer.cfg
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev2.0

[resonance_tester]
accel_chip_x: adxl345 #bedslinger
accel_chip_y: adxl345 #bedslinger
#accel_chip: adx1345 #non-bedslinger
probe_points: 150,150,10
```

closed 2024-03-05 by Jpe230