Jpe230/SonicPad-Debian #42: detailed installation guide

Nocentinia, 2024-01-22:
Hi everyone,
I'm a "newbie" and I have some questions regarding the pre-installation guide.
basically do I have to follow the guide provided by creality and flash their firmware on the sonic pad?
I imagine that this is not the case and instead of mounting the original Creality firmware I should mount another firmware which I cannot find in this guide.
Can someone explain to me better what are the steps to take to be able to install the original klipper on my sonic?
thanks a lot to everyone!!

sfeole, 2024-01-23:
Hey @Nocentinia,  So I'm pretty sure your question is,  "How do I use Crealitys Toolset to Flash my Device"  ??  

https://github.com/CrealityOfficial/Creality_Sonic_Pad_Firmware

Take a look at the link above for starters, if you made if this far  (Finding jpe230's awesome debian port for the sonic pad)  then you should be able to navigate your way around the tooling.

In the Creality Documentation you will find instructions for how to reflash the sonic pad.  Simply follow the guide. When you reach the step to "Burn the firmware to sonic pad"  Use the Debian image provided here:

https://github.com/Jpe230/SonicPad-Debian/releases

Remember to get all 3 files, the t800 z01/z02 and zip file. Then unzip per the instructions.

That's pretty much it.  The sonic pad will reboot into debian and you can follow along with the install guide in the Readme.md. 

Did that answer your question?    Feel free to message me directly if you get stuck or have more questions!

-sfeole

Nocentinia, 2024-01-23:
HI,
thanks for your reply.
I followed the Creality video tutorial to understand the procedure.
the only problem is that I can't find the .img file to upload to the sonic pad.

I tried to download all three files but I don't know how to unzip them.
I only found one file with the .img extension by unzipping the .z01 file using the "The unarchiver" application.
I haven't been able to test if it works yet because I'm waiting for the USB A-USB A cable (because from what I understand a USB C-USB A cable doesn't work)

sfeole, 2024-01-23:
@Nocentinia Hey There,  

I would try to Download 7z for Windows ("I'm assuming you're using that")   https://www.7-zip.org/download.html

Then put all of the files in the same folder and try to extract the .zip

Nocentinia, 2024-01-23:
no, I'm on MacOs.
I tried using 7z but whatever I try to do gives me the directory is non-existent to extract the file.
so I really can't find the image to upload to the Sonic Pad.
Could you help me somehow?
thanks so much for your help

cyclemat, 2024-01-23:
@Nocentinia i can help you come on my discord and we make it https://discord.gg/ZJDftemn

Nocentinia, 2024-01-23:
I don't know why it won't let me log in. I sent you a friend request, please accept it

qwazwsx, 2024-02-09:
I have created a more detailed written tutorial here:

https://github.com/qwazwsx/Advanced-SonicPad/

feel free to PR

Jpe230, 2024-03-05:
> I have created a more detailed written tutorial here:
> 
> https://github.com/qwazwsx/Advanced-SonicPad/
> 
> feel free to PR

I would totally be thankful if you can help me improving the instructions for this repo :)

Blackskyliner, 2024-11-21:
If no one deletes their repositories in the future a link would be already more than enough. Or if @qwazwsx may just PR his work into this repository under INSTALL.md or a newly founded /docs could also benefitial for the project as one whole package. 

As it worked like a charm. Only issue I had was, that I HAD to do all of this from a freshly installed Windows 10 otherwise the driver of the PhoenixSuite never worked for me and never picked up nor installed the driver successfully for the connected device. But may be my long-living windows installations which may be borked at some time in the past for stuff like this.