Jpe230/SonicPad-Debian #45: Host shutdown and reboot from the fluidd menu

sahuddy, 2024-02-18:
Is there a way to enable the host shutdown and reboot buttons in the fluidd ellipsis menu?  
When I click them after a fresh installation, nothing happens.  Thanks in advance!

sahuddy, 2024-02-18:
Please disregard.  The solution is here: https://docs.fluidd.xyz/faq#the-host-reboot--shutdown-commands-dont-work

Brakisshuro, 2024-02-28:
> Please disregard. The solution is here: https://docs.fluidd.xyz/faq#the-host-reboot--shutdown-commands-dont-work

i tried that command and no luck, says command not found.

sahuddy, 2024-02-28:
Hmmm....maybe moonraker needs to be updated?  It worked for me,

Brakisshuro, 2024-02-28:
> Hmmm....maybe moonraker needs to be updated? I'm at v0.8.0-317 and it worked for me,

You copied and pasted That exact command and hit enter? Everything I have is 100% updated

sahuddy, 2024-02-28:
I sure did.

Brakisshuro, 2024-02-28:
> I sure did.

Interesting, okay thank you

Brakisshuro, 2024-02-28:
> I sure did. If it is there though, after changing into the directory, try ./sudo_fix.sh

Are you cd /klipper/ or are you doing it from the root prompt

sahuddy, 2024-02-28:
When connecting, I land in /home/sonic.  From there the provided path works: ./moonraker/scripts/sudo-fix.sh, because I can also cd to ./moonraker/scripts/ run ls (list) and see the sudo-fix.sh script.

Brakisshuro, 2024-02-28:
> When connecting, I land in /home/sonic. From there the provided path works: ./moonraker/scripts/sudo-fix.sh, because I can also cd to ./moonraker/scripts/ run ls (list) and see the sudo-fix.sh script.

I found the script I try to run it and it says command not found. Very weird. I'm in the Moonraker directory right now

sahuddy, 2024-02-28:
From that directory, try: sudo ./sudo-fix.sh

Brakisshuro, 2024-02-28:
> From that directory, try: sudo ./sudo-fix.sh

sonic@tony:~/moonraker/scripts$ sudo ./sudo_fix.sh
This script must not run as root

sahuddy, 2024-02-28:
Ug.  So just: ./sudo_fix.sh then

sahuddy, 2024-02-28:
Cool...sounds like you're good to go.  Have a great night!

closed 2024-03-05 by Jpe230