Jpe230/SonicPad-Debian #47: creality nebula camera

bhast2, 2024-02-25:
Got everything installed and working but I can't get my nebula camera to work. I have tried everything I can think of to make it work. It shows that it is connected, and it sees the camera but can't get video to show. I plug in a different camera, and it works perfect. Any help on this would be great. Thank you

Brakisshuro, 2024-02-28:
I have this issue as well

sahuddy, 2024-02-28:
Try changing the Camera Url Stream to:  http://<your IP address>/webcam/?action=stream and/or the Stream type to MJPEG Stream

Brakisshuro, 2024-02-28:
> Try changing the Camera Url Stream to: http:///webcam/?action=stream and/or the Stream type: to MJPEG Stream

I have tried all the combos I can, when I try to access via the pad, is says "failed to open http://120.0.0.1/webcam/?action=stream"

on fluid, it is just a sliver of the screen.

sahuddy, 2024-02-28:
You are trying to use the system loopback address, which is actually 127.0.0.1, but I meant use the IP address that the pad is using.

Brakisshuro, 2024-02-28:
> You are trying to use the system loopback address, which is actually 127.0.0.1, but I meant use the IP address that the pad is using.

Nginx bad gateway 502

sahuddy, 2024-02-28:
What does the [timelapse] stanza look like in moonraker.conf?

Brakisshuro, 2024-02-28:
> What does the [timelapse] stanza look like in moonraker.conf? Also, are you using crowsnest?

[timelapse]
output_path: /usr/data/printer_data/timelapse/
frame_path: /usr/data/printer_data/frames/
ffmpeg_binary_path: /opt/bin/ffmpeg
snapshoturl: http://localhost:8080/?action=snapshot

sahuddy, 2024-02-28:
Thanks.  That all looks right, so I'm out of ideas.  Sorry.

Brakisshuro, 2024-02-28:
> Thanks. That all looks right, so I'm out of ideas. Sorry.

That's ok, appreciate your help

DDen87, 2024-03-02:
@Jpe230 you know of a fix for the Creality Nebula Camera perhaps? Is there something we can do about this? Thanks in advance for your help!

bhast2, 2024-03-03:
I hope a fix is found soon. What I don't understand is the camera works great on a raspberry pi with the mainsail os on raspberry pi imager.

Jpe230, 2024-03-05:
Is the Nebula cam just an run of the mill webcam? If so I can give it a try to see if I can reproduce with a Logitech cam, otherwise I'll have to save up some money to buy one :p

Brakisshuro, 2024-03-05:
I have some drivers I found if you want to take a look at them. It is creality 's AI camera. It works on stock Sonic pad, and it works on pi installations of Klipper

Brakisshuro, 2024-03-05:
@Jpe230 here is drivers I got from creality after working my way up their foodchain. 
https://drive.google.com/file/d/1zQ3fj2bvAvJTA9mYQD4fU4kFEcTF7Mad/view?usp=sharing

Jpe230, 2024-03-08:
I managed to use the Nebula Camera, all I did was to comment this line:

https://github.com/mainsail-crew/crowsnest/blob/e96cd46ffb4a42ae1fbf23118caa68dd0dac0597/libs/ustreamer.sh#L59

The file is located at /home/sonic/crowsnest/libs/ustreamer.sh

[image]

FWIW my crownest conf is the following:

```
[crowsnest]
log_path: /home/sonic/printer_data/logs/crowsnest.log
log_level: debug                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1280x720                    # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```

DDen87, 2024-03-09:
@Jpe230 as I'm quite new to software and Klipper. How do I get to that file? Do I use terminal (on my Mac)? Or can I reach it from the mainsail frontend of the sonic pad?

bhast2, 2024-03-09:
Here is my ustreamer.sh  best way to edit this is use a program like winscp or something like it

`#!/bin/bash

#### ustreamer library

#### crowsnest - A webcam Service for multiple Cams and Stream Services.
####
#### Written by Stephan Wendel aka KwadFan <me@stephanwe.de>
#### Copyright 2021
#### https://github.com/mainsail-crew/crowsnest
####
#### This File is distributed under GPLv3
####

# shellcheck enable=require-variable-braces

# Exit upon Errors
set -Ee

run_mjpg() {
    local cams
    v4l2_control
    cams="${1}"
    for instance in ${cams} ; do
        run_ustreamer "${instance}" &
    done
    blockyfix
    brokenfocus "${cams}"
    return
}

run_ustreamer() {
    local cam_sec ust_bin dev pt res fps cstm start_param
    cam_sec="${1}"
    # shellcheck disable=SC2153
    ust_bin="${UST_BIN}"
    dev="$(get_param "cam ${cam_sec}" device)"
    pt="$(get_param "cam ${cam_sec}" port)"
    res="$(get_param "cam ${cam_sec}" resolution)"
    fps="$(get_param "cam ${cam_sec}" max_fps)"
    cstm="$(get_param "cam ${cam_sec}" custom_flags 2> /dev/null)"
    noprx="$(get_param "crowsnest" no_proxy 2> /dev/null)"
    # construct start parameter
    if [[ -n "${noprx}" ]] && [[ "${noprx}" = "true" ]]; then
        start_param=( --host 0.0.0.0 -p "${pt}" )
        log_msg "INFO: Set to 'no_proxy' mode! Using 0.0.0.0 !"
    else
        start_param=( --host 127.0.0.1 -p "${pt}" )
    fi

    #Raspicam Workaround
    if [[ "${dev}" = "$(dev_is_legacy)" ]]; then
        start_param+=( -m MJPEG --device-timeout=5 --buffers=3 )
    else
        # Add device
        start_param+=( -d "${dev}" --device-timeout=2 )

        # Use MJPEG Hardware encoder if possible
       # if [ "$(detect_mjpeg "${cam_sec}")" = "1" ]; then
           # start_param+=( -m MJPEG --encoder=HW )
       # fi
    fi

    # set max framerate
    start_param+=( -r "${res}" -f "${fps}" )

    # webroot & allow crossdomain requests
    start_param+=( --allow-origin=\* --static "${BASE_CN_PATH}/ustreamer-www" )
    # Custom Flag Handling (append to defaults)
    if [[ -n "${cstm}" ]]; then
        start_param+=( "${cstm}" )
    fi
    # Log start_param
    log_msg "Starting ustreamer with Device ${dev} ..."
    echo "Parameters: ${start_param[*]}" | \
    log_output "ustreamer [cam ${cam_sec}]"
    # Start ustreamer
    echo "${start_param[*]}" | xargs "${ust_bin}" 2>&1 | \
    log_output "ustreamer [cam ${cam_sec}]"
    # Should not be seen else failed.
    log_msg "ERROR: Start of ustreamer [cam ${cam_sec}] failed!"
}
`

DDen87, 2024-03-09:
@bhast2 Thanks! I'm in. I have copied in your complete file. But still no image. I copied in @Jpe230 his crowsnest.conf.

Can you maybe share your crowsnest.conf? Can you also share you settings of the webcam and on what usb port you have it connected?

Thanks in advance (And of course so far, couldn't have come so far with this without you guys!)!

P.s. I used Cyberduck (if anybody with a Mac is interested.... )

Jpe230, 2024-03-09:
> @bhast2 Thanks! I'm in. I have copied in your complete file. But still no image. I copied in @Jpe230 his crowsnest.conf.
>
>
>
> Can you maybe share your crowsnest.conf? Can you also share you settings [...]

Have you replaced the file under the path I previously mentioned? If so try rebooting the pad or restart the crownest service.

I don't think that the USB port matters, but I'm connecting the cam to the bottom port at the side of the pad.

If you still have problems please share your crownest logs :)

DDen87, 2024-03-09:
@Jpe230 as I wasn't sure I installed everything correctly I reinstalled crows nest, and I saw this come by:
```
Checking if device is a Raspberry Pi ...

Checking if device is a Raspberry Pi ... [FAILED]
This device is not a Raspberry Pi therefore camera-streeamer cannot be installed ...
```
But I'm going the retry and uncomment line 57 in de streamer.sh file in the libs folder. And copy in your crows nest config. See what happens.

Jpe230, 2024-03-09:
Yup that's expected, the pi uses camera-streamer for its backend service, and apparently it is only compatible with Pis, that why we are falling back to ustreamer ;)

DDen87, 2024-03-09:
Hahahaha I just wanted to double check! As I'm not that good with software...

DDen87, 2024-03-09:
@Jpe230 unfortunately did not help. See attached the crows nest log file. Thanks again for taking a look!
[crowsnest.log](https://github.com/Jpe230/SonicPad-Debian/files/14547273/crowsnest.log)

DDen87, 2024-03-09:
[image]

bhast2, 2024-03-09:
> @bhast2 Thanks! I'm in. I have copied in your complete file. But still no image. I copied in @Jpe230 his crowsnest.conf.
>
> Can you maybe share your crowsnest.conf? Can you also share you settings of [...]

I didn't change anything in my crowsnest.conf I just commented out those lines in the ustreamer file and it started working

bhast2, 2024-03-09:
Make sure you have MJPEG STREAM selected.
and use the following for your url info

Camera URL Stream  `/webcam/?action=stream`
Camera URL SnapShot  `/webcam/?action=snapshot`

Brakisshuro, 2024-03-09:
Boom, I got it working!, Thank you gentlemen!

DDen87, 2024-03-10:
@bhast2 Ok, I reinstalled crowsnest again(this time via the repo instead of via kiauh). Changed only the ustreamer.sh file. Added a webcam (the correct URL were already filled in). But still nothing... well something... a white screen...
[image]

bhast2, 2024-03-10:
> @bhast2 Ok, I reinstalled crowsnest again(this time via the repo instead of via kiauh). Changed only the ustreamer.sh file. Added a webcam (the correct URL were already filled in by default). But still nothing... well something... a white screen... ![image](https://private-user-images.githubusercontent.com/152805261/311527496-d8d2a836-6b6d-4784-be98-129e022246f0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAwOTUxNTEsIm5iZiI6MTcxMDA5NDg1MSwicGF0aCI6Ii8xNTI4MDUyNjEvMzExNTI3NDk2LWQ4ZDJhODM2LTZiNmQtNDc4NC1iZTk4LTEyOWUwMjIyNDZmMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxMFQxODIwNTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zYTI1MWViYmJhMWU3ZDFhZWY0NjY1MzhjOTg5YmQ1ZTFmYWM2MDdlYmU5MDMxNTY5NjNiYmI5MGY4YThlZTlkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.HdLkXx6PJdYFuonW_Huv352QN5XD4uYtC01kuWmOCWk)

I installed it thru kiauh. I have the camera plugged into the usb port

bhast2, 2024-03-10:
> @bhast2 Ok, I reinstalled crowsnest again(this time via the repo instead of via kiauh). Changed only the ustreamer.sh file. Added a webcam (the correct URL were already filled in by default). But still nothing... well something... a white screen... !

Make sure you change the config file also

```
[crowsnest]
log_path: /home/sonic/printer_data/logs/crowsnest.log
log_level: debug                      # Valid Options are quiet/verbose/debug
delete_log: false                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
device: /dev/video0                     # See Log for available ...
resolution: 1280x720                    # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```

bhast2, 2024-03-10:
> wont lie, got it working but it has a 6 second delay and fps is BAD so just bought a logitech c270
I don't have lag. here is a video of my stream
[video](https://streamable.com/mdr3yi)

DDen87, 2024-03-10:
@Brakisshuro how you like that cams quality?
@bhast2 I have tried everything... have no idea what's wrong with... probably will have to reinstall the complete Sonic pad. I tried the nebula also on my Mac and it works correctly.

bhast2, 2024-03-10:
> @Brakisshuro how you like that cams quality? @bhast2 I have tried everything... have no idea what's wrong with... probably will have to reinstall the complete Sonic pad. I tried the nebula also on my Mac and it works correctly.

So for me to get it to work I downloaded the newest version of sonic pad Debian.

DDen87, 2024-03-10:
Got a link?

Jpe230, 2024-03-13:
> wont lie, got it working but it has a 6 second delay and fps is BAD so just bought a logitech c270

Maybe poor connection to your router? Mine can achieve 15-20fps stable, with max 1s delay, tbh is quite usable

Brakisshuro, 2024-03-13:
> > wont lie, got it working but it has a 6 second delay and fps is BAD so just bought a logitech c270
> 
> Maybe poor connection to your router? Mine can achieve 15-20fps stable, with max 1s delay, tbh is quite usable

Interesting, mine is hardwired to one of my switches. I tested on Wi-Fi as well and still was getting horrible lag.

pechex, 2024-06-18:
Hi all,

I had this same problem and none of the things posted here helped.
My solution was to change /dev/video0 to /dev/v4l/by-id/usb-UnionImage_Co._Ltd_CCX2F3298_1234567890-video-index0   in crownest.cfg .

I got it from crownest.log or you can see it also on (check screenshot)

```
[crowsnest]
log_path: /home/sonic/printer_data/logs/crowsnest.log
log_level: debug                      # Valid Options are quiet/verbose/debug
delete_log: true                       # Deletes log on every restart, if set to true
no_proxy: false

[cam 1]
mode: ustreamer                         # ustreamer - Provides mjpg and snapshots. (All devices)
                                        # camera-streamer - Provides webrtc, mjpg and snapshots. (rpi + Raspi OS based only)
enable_rtsp: false                      # If camera-streamer is used, this enables also usage of an rtsp server
rtsp_port: 8554                         # Set different ports for each device!
port: 8080                              # HTTP/MJPG Stream/Snapshot Port
#device: /dev/video0                     # See Log for available ...
device: /dev/v4l/by-id/usb-UnionImage_Co._Ltd_CCX2F3298_1234567890-video-index0                     # See Log for available ...
resolution: 1280x720                    # widthxheight format
max_fps: 30                             # If Hardware Supports this it will be forced, otherwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
```

[image]

8Chachie6, 2024-09-17:
Hello everyone I’m having this issue here now my self I was reading through the thread and got a bit confused could some one here help me out ?

8Chachie6, 2024-09-17:
[image]

this is what I get ??

8Chachie6, 2024-09-17:
> @Jpe230 here is drivers I got from another member who worked on creality up their foodchain. https://drive.google.com/file/d/1zQ3fj2bvAvJTA9mYQD4fU4kFEcTF7Mad/view?usp=sharing

This link doesn’t work anymore?

pechex, 2024-09-17:
this->
https://github.com/Jpe230/SonicPad-Debian/issues/47#issuecomment-1986514285

8Chachie6, 2024-10-03:
> @Jpe230 here is drivers I got from another member who worked on creality up their foodchain. https://drive.google.com/file/d/1zQ3fj2bvAvJTA9mYQD4fU4kFEcTF7Mad/view?usp=sharing

this link dosent work... do you still have the drivers ?

8Chachie6, 2024-10-04:
> @Jpe230 I have some drivers I found if you want to take a look at them. It is creality 's AI camera. It works on stock Sonic pad, and it works on pi installations of Klipper

i keep seeing about thees drivers but nun the links to them work??

Lloir, 2024-12-04:
Same problem here

HOWEVER, worked around it Installed ustreamer

then ran ustreamer --host 192.168.1.66 --port 9034

added http://192.168.1.66:9034/stream to camera urlstream and tada

pildorika, 2025-04-23:
A día de hoy sigue existiendo una solución?, ami me sale o error para conectar o se queda en not signal