Jpe230/SonicPad-Debian #50: Unable to use CANbus

oldcarlos, 2024-03-18:
The kernel seems to be compiled without support for CANbus. Haven't been able to use a usb-CAN adapter. I can't seem to find available module to load, either. Would be great to be able to use it for my now all CAN Ender3S1.

Jpe230, 2024-04-11:
There's a branch in this repo with the precompiled kernel with canbus support, you will need to create the flashable image (there's a script under the src folder to create it)

Iirc the branch is old and likely klipperscreen won't work, but if your pad is already running Debian you can flash only the kernel partition using PhoenixSuit.

Drake-W, 2024-04-24:
@Jpe230 
I already have Debian and Klipperscreen up and running, could you explain more on how to flash only the kernel partition?

oldcarlos, 2024-05-12:
I managed to get the canbus kernel working. I had to install a current version, then make an image of the canbus branch using the instructions on the DIY.md file. Using PhoenixSuit, in the Firmware tab, select "Download one or multi partition" I think I then selected BOOTA and BOOT-RESOURCE but not quite sure since I had to try different options. However when I try to use this kernel I get TX error on the network. Eventually the MCU will disconnect. Using a RPi, I get zero error with the same setup. I won't try to look further into the problem, since the kernel is as is and have no alternatives.

bulcke, 2025-04-21:
@Jpe230 Could you add this to the main release or provide a compiled firmware version that includes canbus?