Jpe230/SonicPad-Debian #56: new R818 Repository issue

HoudiniCerini, 2024-05-04:
I'm stuck! 
i flashed everything klipper screen is open and i set up wifi
when i try to sudo apt-get update && sudo apt-get upgrade this is the issue i get
Hit:1 http://deb.debian.org/debian bullseye InRelease
Reading package lists... Done
E: Release file for http://deb.debian.org/debian/dists/bullseye/InRelease is not valid yet (invalid for another 236d 21h 13min 51s). Updates for this repository will not be applied.
Is there a discord for this by chance?

doomedramen, 2024-05-04:
You have to follow the docs where it tells you to set the timezone and then restart. the reason you are getting that error is your system thinks you are in the past.

HoudiniCerini, 2024-05-04:
I didn't see any instructions for that how the heck do I do do that I
watched all of the YouTubes and all...

doomedramen, 2024-05-04:
The front page of this repo.Jpe230/SonicPad-Debian: Port of Debian for the SonicPad (Allwinner R818)github.com

HoudiniCerini, 2024-05-05:
I got it all figured out just gotta get the adxl set up and we will be gtg
🤙🏻

closed 2024-05-14 by Jpe230