Jpe230/SonicPad-Debian #60: Resonance test hangs Klipper

pechex, 2024-07-24:
Hi all,

I recently installed Debian in my Sonic Pad. It went mostly OK but now I´m trying to do the resonance test following  #4  
I all worked fine, except when trying to do the actual test. I use the sensor provided by Creality, which worked fine on the stock firmware.

After running SHAPER_CALIBRATE it goes trough all the frecuencies and at the end it returns a message saying Wait for calculations..
At that point, Klipper is no longer responsive. No command works, restarting it makes ir responsive again, but I get no input shaping.

I also tried TEST_RESONANCES AXIS=X or TEST_RESONANCES AXIS=Y but I get exactly the same results

printer.cfg
```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev2.0

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points:
      117.5,117.5,10
```

Example commands

```
ACCELEROMETER_QUERY 
accelerometer values (x, y, z): -518.144159, 0.000000, 8197.221829
```

```
Wait for calculations..
Testing frequency 133 Hz
Testing frequency 132 Hz
Testing frequency 131 Hz
Testing frequency 130 Hz
Testing frequency 129 Hz
Testing frequency 128 Hz
Testing frequency 127 Hz
<<<< REMOVED>>>>
Testing frequency 17 Hz
Testing frequency 16 Hz
Testing frequency 15 Hz
Testing frequency 14 Hz
Testing frequency 13 Hz
Testing frequency 12 Hz
Testing frequency 11 Hz
Testing frequency 10 Hz
Testing frequency 9 Hz
Testing frequency 8 Hz
Testing frequency 7 Hz
Testing frequency 6 Hz
Testing frequency 5 Hz
Testing axis x
SHAPER_CALIBRATE
```

klippy.log
```
Testing axis x
ADXL345 starting 'adxl345' measurements
toolhead: max_velocity: 300.000000
max_accel: 9333.333000
minimum_cruise_ratio: 0.000000
square_corner_velocity: 5.000000
Testing frequency 5 Hz
[... 23 lines cut ...]
Stats 5393.5: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000034 mcu_task_stddev=0.000022 bytes_write=548910 bytes_read=100491 bytes_retransmit=0 bytes_invalid=0 send_seq=9929 receive_seq=9928 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=14 upcoming_bytes=921 freq=84000710 rpi: mcu_awake=2.317 mcu_task_avg=0.000150 mcu_task_stddev=0.000077 bytes_write=12036 bytes_read=2354192 bytes_retransmit=8 bytes_invalid=0 send_seq=1742 receive_seq=1742 retransmit_seq=918 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000797 adj=50001082  heater_bed: target=0 temp=20.0 pwm=0.000 sysload=1.32 cputime=37.930 memavail=1762888 print_time=2315.111 buffer_time=2.269 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
Testing frequency 130 Hz
Stats 5394.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000034 mcu_task_stddev=0.000022 bytes_write=553599 bytes_read=100992 bytes_retransmit=0 bytes_invalid=0 send_seq=10005 receive_seq=10005 retransmit_seq=0 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=34 upcoming_bytes=986 freq=84000710 rpi: mcu_awake=2.317 mcu_task_avg=0.000150 mcu_task_stddev=0.000077 bytes_write=12112 bytes_read=2373237 bytes_retransmit=8 bytes_invalid=0 send_seq=1753 receive_seq=1753 retransmit_seq=918 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000766 adj=50000575  heater_bed: target=0 temp=20.0 pwm=0.000 sysload=1.38 cputime=38.185 memavail=1762464 print_time=2316.107 buffer_time=2.264 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
Testing frequency 131 Hz
Stats 5395.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=558347 bytes_read=101494 bytes_retransmit=0 bytes_invalid=0 send_seq=10082 receive_seq=10082 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=41 upcoming_bytes=1071 freq=84000709 rpi: mcu_awake=2.317 mcu_task_avg=0.000150 mcu_task_stddev=0.000077 bytes_write=12188 bytes_read=2392222 bytes_retransmit=8 bytes_invalid=0 send_seq=1764 receive_seq=1764 retransmit_seq=918 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000787 adj=50000179  heater_bed: target=0 temp=20.0 pwm=0.000 sysload=1.38 cputime=38.451 memavail=1762284 print_time=2317.111 buffer_time=2.267 print_stall=0 extruder: target=0 temp=20.1 pwm=0.000
Testing frequency 132 Hz
Stats 5396.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=563158 bytes_read=102000 bytes_retransmit=0 bytes_invalid=0 send_seq=10160 receive_seq=10160 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=24 upcoming_bytes=1149 freq=84000708 rpi: mcu_awake=2.317 mcu_task_avg=0.000150 mcu_task_stddev=0.000077 bytes_write=12257 bytes_read=2411243 bytes_retransmit=8 bytes_invalid=0 send_seq=1774 receive_seq=1774 retransmit_seq=918 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000826 adj=50000468  heater_bed: target=0 temp=20.2 pwm=0.000 sysload=1.38 cputime=38.694 memavail=1761600 print_time=2318.111 buffer_time=2.267 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
Testing frequency 133 Hz
Stats 5397.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=567959 bytes_read=102506 bytes_retransmit=0 bytes_invalid=0 send_seq=10238 receive_seq=10238 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=58 upcoming_bytes=1200 freq=84000707 rpi: mcu_awake=2.294 mcu_task_avg=0.000149 mcu_task_stddev=0.000080 bytes_write=12333 bytes_read=2430302 bytes_retransmit=8 bytes_invalid=0 send_seq=1785 receive_seq=1785 retransmit_seq=918 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000793 adj=50000915  heater_bed: target=0 temp=20.0 pwm=0.000 sysload=1.38 cputime=38.955 memavail=1761428 print_time=2319.109 buffer_time=2.264 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
toolhead: max_velocity: 300.000000
max_accel: 4000.000000
minimum_cruise_ratio: 0.500000
square_corner_velocity: 5.000000
Stats 5398.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=569832 bytes_read=102762 bytes_retransmit=0 bytes_invalid=0 send_seq=10269 receive_seq=10269 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=84000705 rpi: mcu_awake=2.294 mcu_task_avg=0.000149 mcu_task_stddev=0.000080 bytes_write=12409 bytes_read=2449285 bytes_retransmit=8 bytes_invalid=0 send_seq=1796 receive_seq=1796 retransmit_seq=918 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000761 adj=50000443  heater_bed: target=0 temp=20.1 pwm=0.000 sysload=1.38 cputime=39.168 memavail=1761268 print_time=2319.209 buffer_time=1.363 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
Stats 5399.6: gcodein=0  mcu: mcu_awake=0.020 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=569838 bytes_read=102883 bytes_retransmit=0 bytes_invalid=0 send_seq=10270 receive_seq=10270 retransmit_seq=0 srtt=0.003 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=84000701 rpi: mcu_awake=2.294 mcu_task_avg=0.000149 mcu_task_stddev=0.000080 bytes_write=12485 bytes_read=2468812 bytes_retransmit=8 bytes_invalid=0 send_seq=1807 receive_seq=1807 retransmit_seq=918 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000776 adj=50000070  heater_bed: target=0 temp=20.1 pwm=0.000 sysload=1.43 cputime=39.450 memavail=1760768 print_time=2319.209 buffer_time=0.335 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
ADXL345 finished 'adxl345' measurements
Stats 5400.6: gcodein=0  mcu: mcu_awake=0.013 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=570180 bytes_read=103049 bytes_retransmit=0 bytes_invalid=0 send_seq=10277 receive_seq=10277 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=84000712 rpi: mcu_awake=2.294 mcu_task_avg=0.000149 mcu_task_stddev=0.000080 bytes_write=12554 bytes_read=2477387 bytes_retransmit=8 bytes_invalid=0 send_seq=1816 receive_seq=1816 retransmit_seq=918 srtt=0.001 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000744 adj=50000271  heater_bed: target=0 temp=20.1 pwm=0.000 sysload=1.43 cputime=39.593 memavail=1739188 print_time=2319.209 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.1 pwm=0.000
Stats 5401.6: gcodein=0  mcu: mcu_awake=0.013 mcu_task_avg=0.000037 mcu_task_stddev=0.000020 bytes_write=570186 bytes_read=103155 bytes_retransmit=0 bytes_invalid=0 send_seq=10278 receive_seq=10278 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=84000709 rpi: mcu_awake=2.294 mcu_task_avg=0.000149 mcu_task_stddev=0.000080 bytes_write=12560 bytes_read=2477402 bytes_retransm [... cut ...] tes=0 upcoming_bytes=0 freq=84000696 rpi: mcu_awake=0.001 mcu_task_avg=0.000010 mcu_task_stddev=0.000011 bytes_write=12596 bytes_read=2477525 bytes_retransmit=8 bytes_invalid=0 send_seq=1823 receive_seq=1823 retransmit_seq=918 srtt=0.000 rttvar=0.000 rto=0.025 ready_bytes=0 upcoming_bytes=0 freq=50000604 adj=49999200  heater_bed: target=0 temp=20.1 pwm=0.000 sysload=1.39 cputime=39.735 memavail=1678468 print_time=2319.209 buffer_time=0.000 print_stall=0 extruder: target=0 temp=20.2 pwm=0.000
```

Blackskyliner, 2024-11-21:
Can you provide your pyenv and all versions of installed packages involved? As it seems to be especially important that the pynum version MUST be set to numpy<1.26 otherwise major errors may occure. Also maybe your observation is fixed in newer versions, I couldcalibrate with my pad + original axdl sensor without any problems.

t-ihmels, 2024-12-07:
I am seeing this also.

Linux SonicPad 4.9.191 #3 SMP PREEMPT Sun Jun 18 03:21:17 UTC 2023 aarch64 GNU/Linux

numpy is 1.19.5

klippy log shows that input_shaping is reenabled at end of test.

If I open the input_shaping panel in KlipperScreen the accelerometer values show up after a second and "Finding ADXL" remains on upper left.

After the panel is opened Web access is lost.  SSH session remains.  Reestart is required to restore functionality.

What pyenv output did you want?

t-ihmels, 2024-12-08:
The reason the input_shaper panel of KlipperScreen hangs is because MEASURE_NOISE_AXES hangs.

21:17 MEASURE_AXES_NOISE
21:17 accelerometer values (x, y, z): 296.082377, 814.226536, 9576.193725
21:17 ACCELEROMETER_QUERY