Jpe230/SonicPad-Debian #62: Klipper doesn't start

JozephSchmoe, 2024-09-05:
That's it. Flashed today successfully following instructions. Klipper repeatedly attempts to start but never does. Can't SSH either but that's because Klipper isn't running, of course.

JozephSchmoe, 2024-09-05:
Matter of fact, while I certainly admire the effort, this entire project feels buggy. Sometimes Moonraker fails to start. Sometimes the sdbus error. Always Klipper fails to start. There isn't much to the setup so I can't imagine anything went "wrong."

Nibeck1309, 2024-09-16:
hello all,

I have the same issue.

do you any idea to mitigate it?

Blackskyliner, 2024-11-22:
As flawless as it works for others I get the feeling there might be different revisions or the Pad has some defects for some...

SSH has nothing to do with Klipper. If the system boots in general AND has connectivity (e.G. wired if not able to configure wireless) You should be able to get the IP from your routers dhcp/client overview and then ssh into it.

If nothing works it sounds more like a flashing problem. Either because of some USB Controller shenanigans or faulty image download. Both of which the flashing software should detect mhm... (I for myself could only successfully flash the firmware in an untouched, freshly installed windows 10 installation)

I for myself flashed it, removed all attachments, booted it and after successfull boot and update process I attached my printer to it, maybe that will work for you?