Jpe230/SonicPad-Debian #64: cant get sonic pads adxl resonance tester

Outoftheshadows3D, 2024-10-30:
okay so i have successfully flashed my sonic pad following your steps, everything's great until i go to do input shaper. i click on input shaper and it tells me on the screen its not configured which it is configured in my printer.cfg. what am i doing wrong or is the resonance tester that came with the pad not compatible ? and if the tester that came with the pad isn't compatible please can this be a fix that gets done, as it would make sense for it to still work...

MDo94, 2024-10-31:
https://www.reddit.com/r/SonicPadCreality/comments/1481s9o/comment/jy8gmck/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

If you google around, the answer you will find.

This is a separate project maintained by someone not related to Creality! First research, then complain. I was as well stuck in many steps but eventually got it working.

nofuturekid, 2024-10-31:
@Outoftheshadows3D Look up the closed issues here and you will find your answer. The original ADXL will work but as for all of the other you have to do additional things, like installing Klipper on the pad itself as a MCU.

But as you have not posted much about your configuration it's just guesswork.

Just follow the usual Klipper instructions for using an ADXL and it will work. Only word of warning, the space on the pad is limited and you will end up with well below 1GB free space left on the device.

Blackskyliner, 2024-11-21:
I think this should either be closed or added to the README (or some docs/ ?), where most is already mentioned.

chaingun427, 2025-01-31:
trying to get the original ADXL working as well, attempting to setup RPi MCU per klipper docs and running intot he following after the "flashing" step: 
Failed to start klipper.service: Unit klipper.service not found.

Have 3 working printers setup on the pad right now, could those existing installs be interfering with the process