Jpe230/SonicPad-Debian #65: USB not working

johnscrub, 2025-01-16:
I have installed , then updated everything. I install multiple instances of klipper. Three to be exact. I follow the USB install guide, but can't get the USB to show up. I have a gcodes for uploaded and it prints. I reboot and USB doesn't show up. Does this only work for one instance. Can somebody share there vegan? Could be fault on my end

qwazwsx, 2025-01-27:
I wrote a little script that uses usbmount to automount any usb device into the `/home/sonic/printer_data/gcodes` folder. For multiple instances you can symlink this to each of your instances gcodes folders.

https://github.com/qwazwsx/SPD-usbmount

I'd love if you could test it out and let me know if it works for you.

johnscrub, 2025-01-27:
> I wrote a little script that uses usbmount to automount any usb device into the `/home/sonic/printer_data/gcodes` folder. For multiple instances you can symlink `/home/sonic/printer_data/gcodes` to each of your instances gcodes folders.
> 
> https://github.com/qwazwsx/SPD-usbmount
> 
> I'd love if you could test it out and let me know if it works for you.

I'm getting a 404 error. I'm willing to try it and give feedback. It would be nice if USB was baked in to omg. Not complaining cause I sure as hell couldn't figure it out. Thanks

qwazwsx, 2025-01-27:
Sorry buddy, looks like I forgot to set the repository to public before sharing. My bad...

If this script works well for people I'll PR it into a new version of SonicPad-Debian. Pending Jpe230's approval.

johnscrub, 2025-01-27:
> Sorry buddy, looks like I forgot to set the repository to public before sharing. My bad...
> 
> If this script works well for people I'll PR it into a new version of SonicPad-Debian. Pending Jpe230's approval.

Thanks just saw this this morning. I will try it this afternoon after work and report back here. Thanks again

johnscrub, 2025-01-28:
> Sorry buddy, looks like I forgot to set the repository to public before sharing. My bad...
> 
> If this script works well for people I'll PR it into a new version of SonicPad-Debian. Pending Jpe230's approval.

I got home and settled. Tried it and this is error I'm getting...

sonic@SonicPad:/tmp$ git clone git@github.com:qwazwsx/SPD-usbmount.git
Cloning into 'SPD-usbmount'...
Warning: Permanently added the ECDSA host key for IP address '140.82.114.3' to the list of known hosts.
[git@github.com](mailto:git@github.com): Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

qwazwsx, 2025-01-28:
This is wanting you to configure an ssh key for git clone. You can use HTTP
instead-- I have a key configured on my pad so I dont get this error.

’git clone
https://github.com/qwazwsx/SPD-usbmount.git’

Sorry for all the silly issues. I pinky promise it's smooth sailing from
here on out.

johnscrub, 2025-01-28:
> This is wanting you to configure an ssh key for git clone. You can use HTTP
> instead-- I have a key configured on my pad so I dont get this error.
>
> ’git clone [...]

All good, just don't ever remember that error before. I'll try and run it again in a few minutes. Thanks for fast response

qwazwsx, 2025-01-28:
Of course! I'm really hoping this works for you! Feel free to open an issue in the SPD-usbmount repo if you have any more issues. I don't want to clutter up this particular issue with chat related to SPD-usbmount.

https://github.com/qwazwsx/SPD-usbmount

By default it will create symlinks to the USB devices in `~/printer_data/gcodes`. Since you have 3 instances you'll probably want to set up some symlinks like this. The README has more info.

```
~/instance1_data/gcodes/USB -> ~/printer_data/gcodes
~/instance2_data/gcodes/USB -> ~/printer_data/gcodes
~/instance3_data/gcodes/USB -> ~/printer_data/gcodes
```

or if you want, you can symlink all the directories together and have one unified gcodes folder -- this is my preferred setup as my gcode files are prefixed with the printer name.

qwazwsx, 2025-01-28:
That's fantastic! I'm glad my code works for you. I've gotten a lot of mail regarding configuring USB so I'm glad I was able to make an easy (ish) method that gets the job done in a quick and dirty way. Once I have a little bit more time I'm going to reflash my SonicPad to reconfigure some things so there should be some documentation changes and clarifications coming in the near future.

johnscrub, 2025-01-28:
> That's fantastic! I'm glad my code works for you. I've gotten a lot of mail regarding configuring USB so I'm glad I was able to make an easy (ish) method that gets the job done in a quick and dirty [...]

Thank you again. If you need a guinea pig to try a new image or test. I'm willing to help. Thanks again. I will leave this open as it seems closed issue didn't show as much in a search.