Jpe230/SonicPad-Debian #69: Screen Rotation?

Aryeh56, 2025-01-31:
Hello, I'm just a hobbyist, so thank you for looking at an amateur question. I am trying to rotate my instance of KlipperScreen because I am using the screen sideways for my "FrankenPrinter"

However, I am having trouble actually finding the screen via SSH to configure. I can see X11 files, so I assumed I'd be able to use "xrandr". However, I'm only getting a screen called "default". That screen does have the correct res and refresh rate, so I assume it's the one, but it returns the following:

> sonic@SonicPad:~$ sudo DISPLAY=:0 xrandr
> xrandr: Failed to get size of gamma for output default
> Screen 0: minimum 1024 x 600, current 1024 x 600, maximum 1024 x 600
> default connected 1024x600+0+0 0mm x 0mm
>    1024x600      60.00*
> sonic@SonicPad:~$ DISPLAY=:0 xrandr --output default --rotate left
> xrandr: output default cannot use rotation "left" reflection "none"

I wanted to check with you on several things. One is that I'm even working in the right place; Is the X11 screen "default" the actual screen on the sonic?

If so, I'd appreciate any guidance on how to work around this error. If not, I would love some guidance on where I can find the config I need to work on.

qwazwsx, 2025-01-31:
See #7 

`/etc/X11/xorg.conf` looks like this by default
```conf
Section "Device"
        Identifier      "Allwinner R818 FBDEV"
        Driver          "fbdev"
        Option          "fbdev" "/dev/fb0"

        Option          "SwapbuffersWait" "true"
EndSection
```

adding `        Option         "Rotate" "CW"` will rotate the display 90 deg clockwise. 

>Enable rotation of the display. The supported values are "CW" (clockwise, 90 degrees), "UD" (upside down, 180 degrees) and "CCW" (counter clockwise, 270 degrees). Implies use of the shadow framebuffer layer. Default: off.
>- https://www.commandlinux.com/man-page/man4/fbdev.4.html

We can then use a transformation matrix to "rotate" the touchscreen inputs

```bash
DISPLAY=:0 xinput set-prop 7 'Coordinate Transformation Matrix'  0 -1 1 1 0 0 0 0 1
```

> See documentation and matrix examples at: https://wiki.ubuntu.com/X/InputCoordinateTransformation

Aryeh56, 2025-01-31:
That did it, of course. Apologies for missing the other thread on the first pass. Will have to figure out the right matrix for my application, but we're in business otherwise. Thank you very much for the speedy help.

closed 2025-01-31 by Aryeh56